Pi4J/pi4j #355: Some classes are not importable

bennymeg, 2024-05-02:
It appears that some of the classes (i.e. DigitalState) are packaged after compilation (.class files)
Which means that they cannot be imported.

In example:
```
// use a configuration builder
var ledConfig = DigitalOutput.newConfigBuilder(pi4j)
        .id("my-dout")
        .name("My LED")
        .address(PIN_LED)
        .shutdown(DigitalState.LOW)
        .initial(DigitalState.HIGH);
```
throws an error: `Cannot resolve symbol 'DigitalState'`.

Is this intentional? Can it be fixed?
Tested with Pi4J v2.5.0 & v2.6.0

taartspi, 2024-05-02:
I added this code to an existing testcase.  Appears to work for me.

my imports include

import com.pi4j.io.gpio.digital.Digital;
import com.pi4j.io.gpio.digital.DigitalInput;
import com.pi4j.io.gpio.digital.DigitalOutput;
import com.pi4j.io.gpio.digital.DigitalState;

 var ledConfig = DigitalOutput.newConfigBuilder(pi4j)
                .id("my-dout")
                .name("My LED")
                .address(18)
                .shutdown(DigitalState.LOW)
                .initial(DigitalState.HIGH);

        var led = pi4j.create(ledConfig);

        logger.info("pin detail " + led);

When executed.
[main] INFO com.pi4j.test.devices.dht22.GpioTest - pin detail @my-dout "My LED" (#18)

Maybe we are not trying this the same way, can you give more details ?

taartspi, 2024-05-02:
Also. with 2.6.0-SNAPSHOT  from my POM file            Note <artifactId>pi4j-v2-devices</artifactId> is mine and not required by you.                                                                                                        .            
    <dependencies>
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-v2-devices</artifactId>
            <version>0.0.1</version>
        </dependency>
        
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-core</artifactId>
            <version>${pi4j.version}</version>
        </dependency>
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-plugin-gpiod</artifactId>
            <version>${pi4j.version}</version>
        </dependency >
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-plugin-pigpio</artifactId>
            <version>${pi4j.version}</version>
        </dependency>
        <dependency>
             <groupId>org.slf4j</groupId>
             <artifactId>slf4j-api</artifactId>
             <version>${slf4j.version}</version>
        </dependency>
        <dependency>
            <groupId>org.slf4j</groupId>
            <artifactId>slf4j-simple</artifactId>
            <version>${slf4j.version}</version>
            
        </dependency>
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-plugin-linuxfs</artifactId>
            <version>${pi4j.version}</version>
        </dependency>
    </dependencies>

bennymeg, 2024-05-02:
IDE: intellij IDEA
Build tool: Gradle 7.2
JVM: 17.0.2

---

Steps to reproduce:
- Create a new Gradle project
- Include the following dependencies:
```
dependencies {
    api group: 'com.pi4j', name: 'pi4j-core', version: '2.6.0'                        
    api group: 'com.pi4j', name: 'pi4j-plugin-raspberrypi', version: '2.6.0'          
    api group: 'com.pi4j', name: 'pi4j-plugin-pigpio', version: '2.6.0'
}
```

- Try to import DigitalState: `import com.pi4j.io.gpio.digital.DigitalState;` 
- An error is thrown: "Cannot resolve symbol 'DigitalState'"

It seems that small subset of the files inside the 'pi4j-core' JAR are compiled.
Here is a partial list of them: `Parity, StopBits, DataBits, FlowControl, DigitalState, PullResistance` ...
These files cannot be imported.

--- minimal test case ---

[image]

taartspi, 2024-05-02:
I followed [Pi4J website > Getting started with Pi4J > Minimal example application](https://pi4j.com/getting-started/minimal-example-application/).

Had intellij open this trusted.
This example had been updated for 2.6.0-SNAPSHOT and newer providers. So I edited it match your details. 

diff --git a/build.gradle b/build.gradl
dependencies {
     implementation 'org.slf4j:slf4j-api:2.0.12'
     implementation 'org.slf4j:slf4j-simple:2.0.12'
-    implementation 'com.pi4j:pi4j-core:2.5.1'
-    implementation 'com.pi4j:pi4j-plugin-raspberrypi:2.5.1'
-    implementation 'com.pi4j:pi4j-plugin-gpiod:2.5.1'
+    implementation 'com.pi4j:pi4j-core:2.6.0'
+    implementation 'com.pi4j:pi4j-plugin-raspberrypi:2.6.0'
+    implementation 'com.pi4j:pi4j-plugin-pigpio:2.6.0'
 }

diff --git a/pom.xml b/pom.xml
index 89238d1..c9b6f4f 100644
--- a/pom.xml
+++ b/pom.xml
@@ -70,7 +70,7 @@
         </dependency>
         <dependency>
             <groupId>com.pi4j</groupId>
-            <artifactId>pi4j-plugin-gpiod</artifactId>
+            <artifactId>pi4j-plugin-pigpio</artifactId>
             <version>${pi4j.version}</version>
         </dependency>
     </dependencies>

b/src/main/java/module-info.javamodule com.pi4j.example {
     // Pi4J MODULES
     requires com.pi4j;
-    requires com.pi4j.plugin.gpiod;
+    requires com.pi4j.plugin.pigpio;
 
     // SLF4J MODULES
     requires org.slf4j;

The import works and the use of DigitalState is good.

When I try to build this ‘sudo ./gradlew build’   I hit an error  General error during semantic analysis: Unsupported class file major version 65.  I have not looked into this but hopefully something specific to my environment.

But I do not see your error.   I am using a new java, as long as you are java 11 or newer that should not be the problem, and at least gradle 6.6

openjdk version "21.0.3" 2024-04-16 LTS

./gradlew -version

------------------------------------------------------------
Gradle 6.6

FDelporte, 2024-05-03:
FYI, minimal example is now bumped to 2.6.0

bennymeg, 2024-05-03:
After a lot of testing, it appears to be a bug in IntelliJ IDEA & Gradle and not in Pi4Jv2.
Upgrading to the latest releases (2024.1 & 8.7 respectively) resolve this issue.