Pi4J/pi4j #360: Unexpected  PI_BAD_HANDLE

SSNC-SushilBohaju, 2024-05-21:
When i close the program, I got the strange error code :  PIGPIO ERROR: PI_BAD_HANDLE; unknown handle
Does anyone know the error on pi4j V2.

taartspi, 2024-05-21:
To try and help more information is needed.  Please provide more details. Pi model, SNAPSHOT used, java version, and the stack trace.   With those someone may have insight to the cause of your exception.

SSNC-SushilBohaju, 2024-05-21:
[image]
[Uploading hs_err_pid1565.log…]()

SSNC-SushilBohaju, 2024-05-22:
> To try and help more information is needed. Please provide more details. Pi model, SNAPSHOT used, java version, and the stack trace. With those someone may have insight to the cause of your exception.

[Err_SnsrCtlApp_20240521.log](https://github.com/Pi4J/pi4j-v2/files/15396486/Err_SnsrCtlApp_20240521.log)

taartspi, 2024-05-22:
1. Are you using Remote Pigpio to execute your code, does the entire program execute on your Pi4 
2. Did the program function correctly until you tried to close the program 
3. Looks like the upload hs_err_pid1565.log was not successful
4. line 1081 shows the pi4j context is already shutdown, which makes sense as you said you closed the program. But the next line 'getGridFeedInOutPowerModbus' implies you still have some portion of the program functioning(maybe a separate thread) .  Is there code still running and it is creating these error messages ?

taartspi, 2024-05-22:
And the error message, most if not all Pigpio I2C  functions can return this Error_Code. Something as simple as you closed the device by it going out of scope,  and then attempted to read or write from the same  device, using a stale handle

SSNC-SushilBohaju, 2024-05-22:
> 1. Are you using Remote Pigpio to execute your code, does the entire program execute on your Pi4
> 2. Did the program function correctly until you tried to close the program
> 3. Looks like the upload [...]

All program running inside rpi
When i close program using cltr+c
[image]
After 1081 line the following logs are generate after re execute the program you can check with time frame
Its just a warning but it make my err log difficult for read so i am looking from where this warn log are printing

taartspi, 2024-05-22:
that NOT INITIALIZED error, if you program was successfully reading from your I2C device  prior to ctrlC I again ask if you have some thread still running after you ctrlC your program. At any rate. The logs yesterday look like some code is repeatedly accessing an I2C device, so I think you need to just use the debugger, set a break point for any of your I2C calls and make sure they are working and see when they begin to fail.

SSNC-SushilBohaju, 2024-05-28:
> that NOT INITIALIZED error, if you program was successfully reading from your I2C device prior to ctrlC I again ask if you have some thread still running after you ctrlC your program. At any rate. [...]

Thank you for your help .... problem is solved. :)

closed 2024-05-28 by SSNC-SushilBohaju

taartspi, 2024-05-28:
Glad you solved it.  Tom