Pi4J/pi4j #361: libpgio 2.6.0 causes core dump

rdratlos, 2024-05-26:
We wanted to migrate a project from a custom LinuxFS based GPIO digital output control implementation to the n**ew Pi4J GpioD plugin**. The reason was that LinuxFS native code uses deprecated kernel API functions. As all tests after update failed, we performed tests using the distributed Minimal Example and **Pi4J v2.6.0**. All attempts failed **with core dump (SIGILL fatal error)** in native code and following error information:

```
Stack: [0xb64b9000,0xb6509000],  sp=0xb6505ef8,  free space=307k
Native frames: (J=compiled Java code, A=aot compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libgpiod13311060481651781708.so+0x21d8]
V  [libjvm.so+0x60e5e0]  os::dll_load(char const*, char*, int)+0x78
V  [libjvm.so+0x48aa48]  JVM_LoadLibrary+0xa4

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  java.lang.ClassLoader$NativeLibrary.load0(Ljava/lang/String;ZZ)Z+0 java.base@11.0.23
j  java.lang.ClassLoader$NativeLibrary.load()Z+56 java.base@11.0.23
j  java.lang.ClassLoader$NativeLibrary.loadLibrary(Ljava/lang/Class;Ljava/lang/String;Z)Z+216 java.base@11.0.23
j  java.lang.ClassLoader.loadLibrary0(Ljava/lang/Class;Ljava/io/File;)Z+46 java.base@11.0.23
j  java.lang.ClassLoader.loadLibrary(Ljava/lang/Class;Ljava/lang/String;Z)V+70 java.base@11.0.23
j  java.lang.Runtime.load0(Ljava/lang/Class;Ljava/lang/String;)V+57 java.base@11.0.23
j  java.lang.System.load(Ljava/lang/String;)V+7 java.base@11.0.23
j  com.pi4j.library.gpiod.util.NativeLibraryLoader.loadLibraryFromClasspath(Ljava/lang/String;)V+229 com.pi4j.library.gpiod@2.6.0
j  com.pi4j.library.gpiod.util.NativeLibraryLoader.load(Ljava/lang/String;Ljava/lang/String;)V+605 com.pi4j.library.gpiod@2.6.0
j  com.pi4j.library.gpiod.internal.GpioD.<clinit>()V+6 com.pi4j.library.gpiod@2.6.0
v  ~StubRoutines::call_stub
j  com.pi4j.library.gpiod.internal.GpioDContext.initialize()V+8 com.pi4j.library.gpiod@2.6.0
j  com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalInputProviderImpl.initialize(Lcom/pi4j/context/Context;)Lcom/pi4j/io/gpio/digital/DigitalInputProvider;+12 com.pi4j.plugin.gpiod@2.6.0
j  com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalInputProviderImpl.initialize(Lcom/pi4j/context/Context;)Ljava/lang/Object;+2 com.pi4j.plugin.gpiod@2.6.0
j  com.pi4j.provider.impl.DefaultRuntimeProviders.initializeProvider(Lcom/pi4j/provider/Provider;)V+59 com.pi4j@2.6.0
j  com.pi4j.provider.impl.DefaultRuntimeProviders.add(Ljava/util/Collection;)Lcom/pi4j/provider/Providers;+125 com.pi4j@2.6.0
j  com.pi4j.provider.impl.DefaultRuntimeProviders.add([Lcom/pi4j/provider/Provider;)Lcom/pi4j/provider/Providers;+5 com.pi4j@2.6.0
j  com.pi4j.provider.impl.DefaultRuntimeProviders.initialize(Ljava/util/Collection;)Lcom/pi4j/provider/impl/RuntimeProviders;+117 com.pi4j@2.6.0
j  com.pi4j.runtime.impl.DefaultRuntime.initialize()Lcom/pi4j/runtime/Runtime;+375 com.pi4j@2.6.0
j  com.pi4j.context.impl.DefaultContext.<init>(Lcom/pi4j/context/ContextConfig;)V+221 com.pi4j@2.6.0
j  com.pi4j.context.impl.DefaultContext.newInstance(Lcom/pi4j/context/ContextConfig;)Lcom/pi4j/context/Context;+5 com.pi4j@2.6.0
j  com.pi4j.context.impl.DefaultContextBuilder.build()Lcom/pi4j/context/Context;+15 com.pi4j@2.6.0
j  com.pi4j.context.impl.DefaultContextBuilder.build()Ljava/lang/Object;+1 com.pi4j@2.6.0
j  com.pi4j.Pi4J.newAutoContext()Lcom/pi4j/context/Context;+18 com.pi4j@2.6.0
j  com.pi4j.example.MinimalExample.main([Ljava/lang/String;)V+27 com.pi4j.example@0.0.1
v  ~StubRoutines::call_stub

siginfo: si_signo: 4 (SIGILL), si_code: 1 (ILL_ILLOPC), si_addr: 0xac7121d8

```

System information:

```
Hardware	: BCM2835
Revision	: 9000c1
Model		: Raspberry Pi Zero W Rev 1.1
model name	: ARMv6-compatible processor rev 7 (v6l)
Features	: half thumb fastmult vfp edsp java tls 

NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
uname:Linux 6.6.28+rpt-rpi-v6 #1 Raspbian 1:6.6.28-1+rpt1 (2024-04-22) armv6l
libc:glibc 2.36 NPTL 2.36 
rlimit (soft/hard): STACK 8192k/infinity , CORE 0k/infinity , NPROC 1292/1292 , NOFILE 1048576/1048576 , AS infinity/infinity , CPU infinity/infinity , DATA infinity/infinity , FSIZE infinity/infinity , MEMLOCK 65536k/65536k

```

Even though GpioD support was introduced with arrival of Raspberry Pi 5 we have not seen any information that other PIs are not supported. 

Any help would be appreciated.

[hs_err_pid6664.log](https://github.com/Pi4J/pi4j-v2/files/15449355/hs_err_pid6664.log)

FDelporte, 2024-05-27:
V2.6 was tested on PIs with ARMv7/8 processors, not on ARMv6, as far as I know. I will need to set up a test to check if I can reproduce this...

FDelporte, 2024-05-27:
@rdratlos it would also be interesting to see what you get out of boardinfo logging, see https://www.pi4j.com/documentation/board-info/

```
console.println("Board model: " + pi4j.boardInfo().getBoardModel().getLabel());
console.println("Operating system: " + pi4j.boardInfo().getOperatingSystem());
console.println("Java versions: " + pi4j.boardInfo().getJavaInfo());

console.println("Board model: " + BoardInfoHelper.current().getBoardModel().getLabel());
console.println("Raspberry Pi model with RP1 chip (Raspberry Pi 5): " + BoardInfoHelper.usesRP1());
console.println("OS is 64-bit: " + BoardInfoHelper.is64bit());
```

rdratlos, 2024-05-27:
```
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected OS: Name: Linux, version: 6.6.28+rpt-rpi-v6, architecture: arm
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected Java: Version: 11.0.23, runtime: 11.0.23+9-LTS, vendor: Azul Systems, Inc., vendor version: Zulu11.72+19-CA
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected board type ZERO_W by code: 9000c1
[main] INFO com.pi4j.context.impl.DefaultContext - Detected board model: Raspberry Pi Zero W

```

@FDelporte 
For BoardInfoHelper I need some more time to fix, as it tries to initialize an auto-context and crashes.

rdratlos, 2024-05-27:
@FDelporte 
Sorry for being late. Please find the missing BoardInfo information in the following:

```
[main] INFO com.pi4j.util.Console - Board model: Raspberry Pi Zero W
[main] INFO com.pi4j.util.Console - Operating system: Name: Linux, version: 6.6.28+rpt-rpi-v6, architecture: arm
[main] INFO com.pi4j.util.Console - Java versions: Version: 11.0.23, runtime: 11.0.23+9-LTS, vendor: Azul Systems, Inc., vendor version: Zulu11.72+19-CA
[main] INFO com.pi4j.util.Console - Board model: Raspberry Pi Zero W
[main] INFO com.pi4j.util.Console - Raspberry Pi model with RP1 chip (Raspberry Pi 5): false
[main] INFO com.pi4j.util.Console - OS is 64-bit: false
[main] INFO com.pi4j.util.Console - JVM memory used (MB): 3.3950881958007812
[main] INFO com.pi4j.util.Console - Board temperature (°C): 47.1

```
This platform is still supported by Raspberry OS Bookworm as well as Zulu Java 11. Would be good to use a "modern" GPIO API and get rid of the deprecated Linux file system based access.

rdratlos, 2024-05-28:
After native build of GpioD libraries on the Pi Zero 1 (ARMv6) itself with hacked POM and build scripts, which force 64 bit ARM arch, we got a working libpgiod.so. autoContext() and subsequent off/on of the digital output worked. Seems that something goes wrong, when building the libraries using cross-compiler on the 64 bit production machine.

Btw.: Native GPIO builds on this slow PI has become inexpensive and works pretty fast. Nothing compared to the nightmare with LinuxFS native builds two years ago.

FDelporte, 2024-05-28:
Hi @rdratlos great to hear you could solve the problem and performance is a lot better!
Can you please share the changes that need to be done so we can try to get them into the build scripts? 
Thanks a lot in advance!

rdratlos, 2024-05-28:
Hi @FDelporte,
at least I could prove that GpioD library native build still works on 32-bit ARMv6. The more generic hack is now as follows:

```
diff --git a/libraries/pi4j-library-gpiod/src/main/native/build.sh b/libraries/pi4j-library-gpiod/src/main/native/build.sh
index 64ee5b0a..6011850d 100755
--- a/libraries/pi4j-library-gpiod/src/main/native/build.sh
+++ b/libraries/pi4j-library-gpiod/src/main/native/build.sh
@@ -51,9 +51,9 @@ fi
 
 # validate compatible CPU architecture
 ARCHITECTURE=$(uname -m)
-if [[ (("$ARCHITECTURE" != "aarch64") && ("$ARCHITECTURE" != "amd64") && ("$ARCHITECTURE" != "x86_64")) ]]; then
-    echo "This native build is only supported on Linux-based systems running on an Intel/AMD or ARM 64-bit platform."
-    echo "BUILD ABORTED; REASON: ARCHITECTURE='$ARCHITECTURE'; EXPECTED='aarch64|amd64|x86_64'"
+if [[ (("$ARCHITECTURE" != "aarch64") && ("$ARCHITECTURE" != "armv6l") && ("$ARCHITECTURE" != "amd64") && ("$ARCHITECTURE" != "x86_64")) ]]; then
+    echo "This native build is only supported on Linux-based systems running on an Intel/AMD or ARM platform."
+    echo "BUILD ABORTED; REASON: ARCHITECTURE='$ARCHITECTURE'; EXPECTED='aarch64|armv6l|amd64|x86_64'"
     exit 1
 fi
 
@@ -78,7 +78,7 @@ if [[ "${PI4J_BUILDER}" != "" ]]; then
    echo "Running inside a Pi4J Docker Builder image; [version=${PI4J_BUILDER}; arch=${PI4J_BUILDER_ARCH}]"
    echo "No need to check or install build environment prerequisites."
 else
-   # if this is a Linux-based system and a 64-bit Intel/AMD or ARM platform, then we can install the prerequisites
+   # if this is a Linux-based system and a Intel/AMD or ARM platform, then we can install the prerequisites
    # download and install development prerequisites
    ./build-prerequisites.sh
 fi

```

I fear that we have now working 32 bit libraries but broken 64 libs. I have attached a zip with a 2.6.0 Maven cache, which contains all native libs build on the Pi Zero. Would be interesting to know, if the aarch64 work on later 64 bit Pi.

[pi4j-library-gpiod-2.6.0_mvn.zip](https://github.com/Pi4J/pi4j-v2/files/15465500/pi4j-library-gpiod-2.6.0_mvn.zip)

FDelporte, 2024-05-28:
@eitch is this something you can look into regarding the build changes?

According to @taartspi could be related to https://github.com/Pi4J/pi4j-v2/issues/354, https://github.com/Pi4J/pi4j-v2/issues/217, https://github.com/Pi4J/pi4j-v2/issues/28

rdratlos, 2024-05-31:
@eitch:
It makes not much sense to re-enable ARMv6 native builds. These Pis are way to slow. I believe that there is an issue with the GPIO library POMs. When I download 2.6.0 and Maven build Pi4J I get the core dump. After rebuilding the native GpioD library on a Pi Zero 1 (ARMv6) using

`mvn clean install -Pnative`

with hacked build script (see above) to enable ARMv6 builds everything works fine again.
But if I then rebuild the complete Pi4J suite using

`mvn clean install`

the minimal example app crashes again with **_java.lang.UnsatisfiedLinkError_** error:
```
[main] INFO com.pi4j.Pi4J - New auto context                                                                                                                                   
[main] INFO com.pi4j.Pi4J - New context builder                                                                                                                                
[main] TRACE com.pi4j.context.impl.DefaultContextBuilder - invoked 'build()'                                                                                                   
[main] TRACE com.pi4j.context.impl.DefaultContext - new Pi4J runtime context initialized [config=com.pi4j.context.impl.DefaultContextBuilder$1@173813a]                        
[main] DEBUG com.pi4j.runtime.impl.DefaultRuntime - Pi4J runtime context successfully created & initialized.'                                                                  
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected OS: Name: Linux, version: 6.6.28+rpt-rpi-v6, architecture: arm                                                  
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected Java: Version: 11.0.23, runtime: 11.0.23+9-LTS, vendor: Azul Systems, Inc., vendor version: Zulu11.72+19-CA     
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected board type ZERO_W by code: 9000c1                                                                               
[main] INFO com.pi4j.context.impl.DefaultContext - Detected board model: Raspberry Pi Zero W                                                                                   
[main] INFO com.pi4j.context.impl.DefaultContext - Running on: Name: Linux, version: 6.6.28+rpt-rpi-v6, architecture: arm                                                      
[main] INFO com.pi4j.context.impl.DefaultContext - With Java version: Version: 11.0.23, runtime: 11.0.23+9-LTS, vendor: Azul Systems, Inc., vendor version: Zulu11.72+19-CA    
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Initializing Pi4J context/runtime...                                                                                        
[main] TRACE com.pi4j.runtime.impl.DefaultRuntime - detected plugin: [com.pi4j.plugin.gpiod.GpioDPlugin] in classpath; calling 'initialize()'                                  
[main] TRACE com.pi4j.runtime.impl.DefaultRuntime - detected plugin: [com.pi4j.plugin.raspberrypi.RaspberryPiPlugin] in classpath; calling 'initialize()'                      
[main] WARN com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_INPUT RaspberryPi Digital Input (GPIO) Provider with priority 0 as lower priority than GpioD Digit
al Input (GPIO) Provider which has priority 150                                                                                                                                
[main] WARN com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_OUTPUT RaspberryPi Digital Output (GPIO) Provider with priority 0 as lower priority than GpioD Dig
ital Output (GPIO) Provider which has priority 150                                                                                                                             
[main] TRACE com.pi4j.provider.impl.DefaultRuntimeProviders - adding providers: [count=6]                                                                                      
[main] TRACE com.pi4j.provider.impl.DefaultRuntimeProviders - adding provider: [id=raspberrypi-serial; name=RaspberryPi Serial Provider; class=com.pi4j.plugin.raspberrypi.prov
ider.serial.RpiSerialProviderImpl]                                                                                                                                             
[main] TRACE com.pi4j.provider.impl.DefaultRuntimeProviders - invoked 'add()' provider [count=1]                                                                               
[main] TRACE com.pi4j.provider.impl.DefaultRuntimeProviders - adding provider to managed io map [id=raspberrypi-serial; name=RaspberryPi Serial Provider; class=com.pi4j.plugin
.raspberrypi.provider.serial.RpiSerialProviderImpl]                                                                                                                            
[main] TRACE com.pi4j.provider.impl.DefaultRuntimeProviders - initializing provider [id=raspberrypi-serial; name=RaspberryPi Serial Provider; class=com.pi4j.plugin.raspberrypi
.provider.serial.RpiSerialProviderImpl]                                                                                                                                        
[main] DEBUG com.pi4j.provider.impl.DefaultRuntimeProviders - added io to managed provider map [id=raspberrypi-serial; name=RaspberryPi Serial Provider; class=com.pi4j.plugin.
[... cut ...]
        at com.pi4j@2.6.3-SNAPSHOT/com.pi4j.Pi4J.newAutoContext(Pi4J.java:73)
        at com.pi4j.example@0.0.1-SNAPSHOT/com.pi4j.example.MinimalExample.main(MinimalExample.java:92)

```
If I rebuild the native library again or just overwrite the local Maven cache with the previous rebuild above, the minimal example works fine again. Downloading the native libraries from Maven repository and then uploading again seems to break things. Never experienced this with LinuxFS native libraries, which are too large to native build on a Pi Zero 1.

FDelporte, 2024-08-05:
Side info regarding supported JDK versions on ARMv6: a short time ago, I tried out all versions listed by SDKMAN, which "should" be compatible with ARMv6. Turns out only Azul Zulu 8 and 11 do really work. No newer versions, and no versions of Liberica. There are plans to re-introduce 32bit support in future versions of OpenJDK as these platforms are still used a lot and have some very specific use-cases.

hackerjimbo, 2024-08-05:
I think the solution is simpler than we think. However, I'm struggling to find the build commands for the 32-bit binary libraries. I suspect a simple `--with-arch=armv6` flag may resolve the problem.

Can someone point me to the location of the build image?

eitch, 2024-08-05:
@hackerjimbo This is the image: https://github.com/Pi4J/pi4j-docker

hackerjimbo, 2024-08-05:
Well I'll be jiggered! It turns out that ubuntu on 32-bit ARM compiles to thumb mode _by default_! This happens to work on the v7+ Pies because the hardware supports it. However, that _doesn't_ work on v6 because the thumb mode (1) doesn't have floating point.

_All_ 32-bit Pies operate in ARM mode not thumb mode. Backwards compatibility, don't you know…

hackerjimbo, 2024-08-05:
So, I hear you ask, how does the ARM processor know if it's in thumb or ARM mode?

Well, thumb instructions are 16-bit and 2 byte aligned. ARM instructions are 32-bit and 4 byte aligned. So those crafty chaps in Cambridge used the bottom bit of the PC to determine the processor mode: 0 → ARM and 1 → thumb. Clever, but _nasty_!

hackerjimbo, 2024-08-06:
Status update: I successfully built the `.o` files with the correct architecture. _However_, when it links to form the `.so`  _somehow_ it wants to add some v7 code in as well. Most annoying!

hackerjimbo, 2024-08-07:
Pondering this there may be another issue. I would _hope_ that the java 11 from OpenJDK for v7 is compatible API-wise with the Azul JDK 11, but what if it's not?

I suspect we may need to supply a “patch” to modify the jar file so that it has v6/Azul  `.so` files for Pi 0/1s.

I disagree with the comment above that Pi 0/1s don't have the CPU power. As I sit here now I have 2 Pi 1s and 2 Pi 0s in the room performing everything from CO₂ monitoring to a clock on a 128×64 LCD screen. The CPUs are barely used.

karsteny, 2024-08-10:
Hey, I've got this exact issue after attempting to upgrade from an ancient version, 2.1.1, because that version that has issue #244 (can't open multiple serial/SPI), but it was only fixed in 2.6.0, which has the new Gpio provider that doesn't work on the Zero W.

I'm pretty inexperienced when it comes to hacking the original code, right now I just use the dependency with Maven and I am unaware of how to do the build script modification that is mentioned earlier. Can anyone give me some pointers on how to do that (or, patch the fix from #244 into the older version, if that's easier than modifying the build scripts)?

hackerjimbo, 2024-08-14:
Sorry for the delay in responding, but I've been on holiday.

I think I'm going to have to provide a patch for the pi4j jar. The only non-difficult way to build the libraries for Pi 0/1s is to build it on a 32-bit Raspberry Pi OS (though that doesn't have to be on a 0/1, you could build it on a 5 — but it _must_ be 32-bit). I'll see what I can do.

hackerjimbo, 2024-09-28:
Well that was a long wait from me! However, I've just had Pi 2 die on me (odd fault, everything seems fine except that the USB appears to be dead) so I now need this is use a replacement 1B+ (which I have) rather than get a new 4/5 or reuse a needlessly power 3 that I want for something else.

I plan to initially just patch `lib/armhf/libpi4j-pigpio.so` in `pi4j-library-pigpio.jar` or should I also/alternatively patch `pi4j-plugin-gpiod.jar`? What's the default and what's the best chocie?

Looking at `pi4j-v2/libraries/pi4j-library-pigpio/src/main/native/Makefile` it seems we're building against Java 11. Is this correct?

hackerjimbo, 2024-10-01:
So I tried building against the system version of libgpiod rather than the custom and old one pi4j uses. This seems to be a better idea as _someone else_ maintains it. Is this a good idea?

It didn't work, I ended up with an `Exception in thread "main" java.lang.UnsatisfiedLinkError: 'java.lang.Long com.pi4j.library.gpiod.internal.GpioD.gpiod_chip_iter_new()'` error (which wasn't very helpful).

Thoughts?

eitch, 2024-10-03:
@hackerjimbo the gpiod_chip_iter_new() is a native method, which is provided not by libgpiod, but by pi4j-library-gpiod. So maybe this library isn't available? Do you have more of the stack trace?

hackerjimbo, 2024-10-03:
Here's the full output of the run. I'm wondering if the issue is it loading the pi4j-library-gpiod jar _or_ if the binary library that it seems to extract has undefined references. I'm investigating more…

```
jim@pi32:~/Build/pi4j-example-minimal/target/distribution $ java --module-path . --module com.pi4j.example/com.pi4j.example.MinimalExample
[main] INFO com.pi4j.util.Console - 

[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - 
[main] INFO com.pi4j.util.Console -                   <-- The Pi4J Project -->                  
[main] INFO com.pi4j.util.Console -                    Minimal Example project                  
[main] INFO com.pi4j.util.Console - 
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - 
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected OS: Name: Linux, version: 6.6.47+rpt-rpi-v7, architecture: arm
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected Java: Version: 17.0.12, runtime: 17.0.12+7-Raspbian-2deb12u1rpt1, vendor: Raspbian, vendor version: null
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected board type MODEL_3_B_PLUS by code: a020d3
[main] INFO com.pi4j.context.impl.DefaultContext - Detected board model: Raspberry Pi 3 Model B+
[main] INFO com.pi4j.context.impl.DefaultContext - Running on: Name: Linux, version: 6.6.47+rpt-rpi-v7, architecture: arm
[main] INFO com.pi4j.context.impl.DefaultContext - With Java version: Version: 17.0.12, runtime: 17.0.12+7-Raspbian-2deb12u1rpt1, vendor: Raspbian, vendor version: null
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Initializing Pi4J context/runtime...
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_OUTPUT RaspberryPi Digital Output (GPIO) Provider with priority 0 as lower priority than GpioD Digital Output (GPIO) Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_INPUT RaspberryPi Digital Input (GPIO) Provider with priority 0 as lower priority than GpioD Digital Input (GPIO) Provider which has priority 150
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'java.lang.Long com.pi4j.library.gpiod.internal.GpioD.gpiod_chip_iter_new()'
	at com.pi4j.library.gpiod@2.7.0-SNAPSHOT/com.pi4j.library.gpiod.internal.GpioD.gpiod_chip_iter_new(Native Method)
	at com.pi4j.library.gpiod@2.7.0-SNAPSHOT/com.pi4j.library.gpiod.internal.GpioD.chipIterNew(GpioD.java:342)
	at com.pi4j.library.gpiod@2.7.0-SNAPSHOT/com.pi4j.library.gpiod.internal.GpioDContext.initialize(GpioDContext.java:47)
	at com.pi4j.plugin.gpiod@2.7.0-SNAPSHOT/com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalOutputProviderImpl.initialize(GpioDDigitalOutputProviderImpl.java:78)
	at com.pi4j.plugin.gpiod@2.7.0-SNAPSHOT/com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalOutputProviderImpl.initialize(GpioDDigitalOutputProviderImpl.java:47)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.provider.impl.DefaultRuntimeProviders.initializeProvider(DefaultRuntimeProviders.java:276)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.provider.impl.DefaultRuntimeProviders.add(DefaultRuntimeProviders.java:252)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.provider.impl.DefaultRuntimeProviders.add(DefaultRuntimeProviders.java:232)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.provider.impl.DefaultRuntimeProviders.initialize(DefaultRuntimeProviders.java:357)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.runtime.impl.DefaultRuntime.initialize(DefaultRuntime.java:318)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.context.impl.DefaultContext.<init>(DefaultContext.java:113)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.context.impl.DefaultContext.newInstance(DefaultContext.java:76)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.context.impl.DefaultContextBuilder.build(DefaultContextBuilder.java:309)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.context.impl.DefaultContextBuilder.build(DefaultContextBuilder.java:49)
	at com.pi4j@2.7.0-SNAPSHOT/com.pi4j.Pi4J.newAutoContext(Pi4J.java:70)
	at com.pi4j.example@0.0.1/com.pi4j.example.MinimalExample.main(MinimalExample.java:91)
```

hackerjimbo, 2024-10-03:
Running using  `strace -f -e openat java --module-path . --module com.pi4j.example/com.pi4j.example.MinimalExample` I get the following (extract) so it seems that it _is_ creating the shared libraries.

```
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_INPUT RaspberryPi Digital Input (GPIO) Provider with priority 0 as lower priority than GpioD Digital Input (GPIO) Provider which has priority 150
[pid 10355] openat(AT_FDCWD, "/home/jim/Build/pi4j-example-minimal/target/distribution/./pi4j-plugin-gpiod.jar", O_RDONLY|O_LARGEFILE) = 13
[pid 10355] openat(AT_FDCWD, "/home/jim/Build/pi4j-example-minimal/target/distribution/./pi4j-plugin-raspberrypi.jar", O_RDONLY|O_LARGEFILE) = 13
[pid 10355] openat(AT_FDCWD, "/usr/lib/jvm/java-17-openjdk-armhf/conf/security/java.security", O_RDONLY|O_LARGEFILE) = 13
[pid 10355] openat(AT_FDCWD, "/dev/random", O_RDONLY|O_LARGEFILE) = 13
[pid 10355] openat(AT_FDCWD, "/dev/urandom", O_RDONLY|O_LARGEFILE) = 14
[pid 10355] openat(AT_FDCWD, "/tmp/libgpiod12111534370470057694.so", O_WRONLY|O_CREAT|O_EXCL|O_LARGEFILE, 0600) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libgpiod12111534370470057694.so", O_WRONLY|O_CREAT|O_EXCL|O_LARGEFILE, 0666) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libgpiod12111534370470057694.so", O_RDONLY) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libgpiod12111534370470057694.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libpi4j-gpiod2223814191735095063.so", O_WRONLY|O_CREAT|O_EXCL|O_LARGEFILE, 0600) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libpi4j-gpiod2223814191735095063.so", O_WRONLY|O_CREAT|O_EXCL|O_LARGEFILE, 0666) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libpi4j-gpiod2223814191735095063.so", O_RDONLY) = 15
[pid 10355] openat(AT_FDCWD, "/tmp/libpi4j-gpiod2223814191735095063.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 15
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'java.lang.Long com.pi4j.library.gpiod.internal.GpioD.gpiod_chip_iter_new()'
	at com.pi4j.library.gpiod@2.7.0-SNAPSHOT/com.pi4j.library.gpiod.internal.GpioD.gpiod_chip_iter_new(Native Method)
	at com.pi4j.library.gpiod@2.7.0-SNAPSHOT/com.pi4j.library.gpiod.internal.GpioD.chipIterNew(GpioD.java:342)
```

hackerjimbo, 2024-10-03:
I just tried building pi4j on a fully up-to-date 32-bit Raspberry Pi OS 12 (bookworm) running on a Pi 3. It failed with the same error (see issue #387 ). 

The reason I'm trying to build on that configuration is that I want to create the shared objects that run on a Pi 0 and 1. The 32-bit compiler does this.

The problem is that you can't build 32-bit shared libraries on 64-bit machine nor vice versa. But if we can build just those we can place them in the shared library download area and they get automatically included.

I think having it work on Pi 1s and 0s (especially 0s) is important because they're used a lot in embedded systems (not least by me!) The size and cheapness (of the 0s) is such a winner.