Pi4J/pi4j #369: WARN noise in the log about Ignoring providers on every startup

ylexus, 2024-07-26:
I have both gpiod (I need stable digital input and output) and linuxfs (I need i2c) plugins in my classpath. Both support digital i/o, so this results in
```
2024-07-26 23:26:35,286 WARN [main] c.p.r.i.DefaultRuntime Ignoring provider DIGITAL_INPUT LinuxFS Digital Input (GPIO) Provider with priority 100 as lower priority than GpioD Digital Input (GPIO) Provider which has priority 150
```
I am sure it's a valid use case for a user to want some providers from one plugin and some from another. So it's not a problem, but a BAU. 

In my application I have zero ERROR/WARN logging approach, so any ERROR or WARN is emailed to me. So on every startup I am receiving emails with the messages above.

Can the severity of this and similar messages be reduced to INFO?

ylexus, 2024-07-27:
Another example of what seems to should have been an INFO, if not DEBUG level message. Any reason it's a WARN?
```
2024-07-27 13:26:47,405 WARN [pi4j-shutdown] c.p.r.i.DefaultRuntimeRegistry Shutting down of IO DIN-25 took 24ms
```

FDelporte, 2024-07-27:
Correct remarks @ylexus. Log levels deserve a review. Do you happen to have some time to propose a merge request? Otherwise I will take a look in September after the holidays.

ylexus, 2024-07-27:
> Correct remarks @ylexus. Log levels deserve a review. Do you happen to have some time to propose a merge request? Otherwise I will take a look in September after the holidays.

Done, also done semi-automated removal of string concatenation in logging, except places where the thread safety of logging arguments is not certain.