Pi4J/pi4j #383: Increase amount of write bytes to SPI devices over 65535

mores, 2024-09-16:
When trying to send large amounts of data to SPI device, users will see:
`2024-09-15 17:24:14,183 WARN [PiGpioNativeImpl:263] () PIGPIO ERROR: PI_BAD_SPI_COUNT; bad SPI count`

This is the largest buffer size this library will allow:
```
#define PI_MAX_SPI_DEVICE_COUNT (1<<16)
https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpio.h#L726

   if (count > PI_MAX_SPI_DEVICE_COUNT)
      SOFT_ERROR(PI_BAD_SPI_COUNT, "bad count (%d)", count);
https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpio.c#L4854C1-L4855C61
```

Break up large data into smaller chunks that can be sent to the SPI device.

FDelporte, 2024-09-16:
See discussion #377: in https://github.com/Pi4J/pi4j-v2/discussions/377

eitch, 2024-09-17:
Fix applied through merge #384

closed 2024-09-17 by eitch