Pi4J/pi4j #385: Pi4j OS

jtb910, 2024-09-23:
I've used the files in both Pi4J-Basic-OS-0.3.0.img.zip and Pi4J-Basic-OS-main.img.zip. I've verifying they are good via the SHA256 checksums. I've used multiple SDCards created with the Raspberry Pi Imager. I've tried both a model 2 and a model 5 raspberry pi and haven't been successful. The model 2 stays on the color bar initial screen and the model 5 gives me the following screen (attached pic). I have also tried changing the os-check value as suggested but then it just gives me a blank screen. Any suggestions??
Thanks jb!
[IMG_1259z - Copy](https://github.com/user-attachments/assets/dd46b484-8570-4ed3-aa5d-65387ba0f5f3)

eitch, 2024-10-03:
Hi, the Pi4j OS doesn't work on the Raspberry Pi 5. On a Raspberry Pi 5, just do a standard install, as described here: https://www.raspberrypi.com/software/

After starting the Raspberry Pi, you can install Java and everything you need to build pi4j on the Raspberry Pi, using the following [script](https://github.com/eitch/pi4j-test/blob/develop/src/assembly/setup.sh):

    #!/bin/bash -e
    sudo raspi-config nonint do_i2c 0
    sudo raspi-config nonint do_ssh 0
    sudo raspi-config nonint do_serial_hw 0
    sudo raspi-config nonint do_serial_cons 1
    sudo raspi-config nonint do_onewire 0
    sudo systemctl disable hciuart
    echo "dtoverlay=disable-bt" | sudo tee -a /boot/firmware/config.txt
    
    sudo apt install -y i2c-tools vim git java-common libxi6 libxrender1 libxtst6
    curl -s "https://get.sdkman.io" | bash
    source .bashrc
    sdk install maven
    
    mkdir -p ~/Downloads
    cd ~/Downloads
    wget https://cdn.azul.com/zulu/bin/zulu21.34.19-ca-jdk21.0.3-linux_arm64.deb
    sudo dpkg -i zulu21.34.19-ca-jdk21.0.3-linux_arm64.deb

You can execute the script with the following command:

    curl -s "https://raw.githubusercontent.com/eitch/pi4j-test/develop/src/assembly/setup.sh" | bash

Also for the Raspberry Pi 2, it's probably better if you also use the standard Raspberry Pi OS, and try this script. But then you'll need to check if you are running a 64-bit or 32-bit architecture

closed 2024-10-03 by eitch

FDelporte, 2024-10-03:
@eitch thanks for the explanation and script. I added it to https://www.pi4j.com/pi4j-os/