Pi4J/pi4j #388: I2C writeRead method

kasperski12, 2024-09-30:
Hi all
I think that there is bug in default implementation of writeRead method in I2c.java
`default int writeRead(byte[] writeBuffer, int writeSize, int writeOffset, byte[] readBuffer, int readSize,
        int readOffset) {
        return execute(() -> {
            int written = write(writeBuffer, writeOffset, writeSize);
            if (**written != writeOffset**)
                throw new IllegalStateException(
                    "Expected to write " + writeOffset + " bytes but only wrote " + written + " bytes");
            return read(readBuffer, readOffset, readSize);
        });
    }`
    The if conditions compares written to writeOffset instead of writeSize which is the number of bytes to write.

eitch, 2024-10-03:
Thanks, good catch! There really was wrong! It will be fixed in the next release!

FDelporte, 2024-10-04:
Included in 2.7.0

closed 2024-10-04 by FDelporte