Pi4J/pi4j #393: Unknown board "Raspberry Pi Compute Module 4"

rkopriva, 2024-10-13:
I'm trying to use the library on EDATec Industrial device, which identifies itself as "Raspberry Pi Compute Module 4". This results to unknown board, because the BoardModel.java is only looking for board name that starts with "Compute Module 4".
Can I specify custom board name?

FDelporte, 2024-10-14:
Hi @rkopriva. Boards are not detected based on their name.

Can you please check the output of the following commands:

```
# 1. hard coded board version number
cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'

# 2. board name if first one fails
cat /proc/device-tree/model
```

The change you proposed in #394 would cause a conflict as it repeats the `List<String> boardCodes` of `COMPUTE_4`. We may find a workaround based on the output of the two commands above. But please be aware, at this point, that the Pi4J V2 project only aims to support the Raspberry Pi, so we are not aware of how it works on other boards.

rudolfkopriva, 2024-10-14:
The board version number is d03141. I see it was added in version 2.7.0. I've updated the lib to the latest version and it started working. Thank you.

FDelporte, 2024-10-14:
That's great @rudolfkopriva. Indeed I discovered myself "the hard way" that there are two ranges of board versions for the Compute 4... ;-)

closed 2024-10-14 by FDelporte