Pi4J/pi4j #406: Old boards not auto detected.

cniesen, 2024-11-03:
I have an old model B+ board.  The BoardInfoHelper doesn't detect this board because it returns the old code of 0010 at

https://github.com/Pi4J/pi4j-v2/blob/7b4e52666cd97ea9b62da00a5be29b4c86b69ac7/pi4j-core/src/main/java/com/pi4j/boardinfo/util/BoardInfoHelper.java#L32

In the BoardModel this code isn't listed for the board.

https://github.com/Pi4J/pi4j-v2/blob/7b4e52666cd97ea9b62da00a5be29b4c86b69ac7/pi4j-core/src/main/java/com/pi4j/boardinfo/definition/BoardModel.java#L65

Please add the old board codes.  Here's a reference for these codes.  Thanks.

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/revision-codes.adoc

cniesen, 2024-11-03:
Additional info:
```
$ cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'
0010

$ cat /proc/device-tree/model
Raspberry Pi Model B Plus Rev 1.2
```

FDelporte, 2024-11-04:
Thanks @cniesen, very usefull link! I was indeed missing the codes for those old boards...

Can you please review the changes in https://github.com/Pi4J/pi4j-v2/pull/407?

FDelporte, 2024-11-04:
And thanks @cniesen for adding the additional info with the output of the `cat` command!

cniesen, 2024-11-05:
> Thanks @cniesen, very usefull link! I was indeed missing the codes for those old boards...
> 
> Can you please review the changes in #407?

@FDelporte The PR #407 looks good to me.  Thanks.

cc @eitch