Pi4J/pi4j #411: UnsatisfiedLinkError libpi4j-pigpio.so

RonaldTHerman, 2024-11-15:
Yeah, I know. This is a beginners problem except I've been using pi4j for several years now.. I have two java apps running on separate Pi's. (pi4b and pi3b) two days ago everything was running fine then all at once I get the old **UnsatisfiedLinkError** for **libpi4j-pigpio**. Both apps are developed on MacOS using NetBeans 12.6 so my first suspicion was the common jar files that both use. Not knowing if a file was corrupted, I upgraded to pi4j- 2.7 in NetBeans. That didn't work.. I triple checked my scrip files that set up the environment to make sure the parameters "-Dpi4j.library.path=$CLASSPATH -Djava.lib.path=$CLASSPATH" are in the command line.That checked.. The only other thing that I did was system upgrades on both Pi's.. I see the java I'm running with now is dated Oct 2024... since it's been a while since I did an upgrade I'm wondering if there is something fishy the latest version. QUESTION:is there a way to get an older version of java and how do I remove the newer and replace with the older version?

FDelporte, 2024-11-16:
It would be very suspicious if the Java update breaks Pi4J. Can you provide the full output of `java -version` please?

RonaldTHerman, 2024-11-16:
I also forgot to mention that I tried running an earlier version of my code to no avail. 

Here's my evidence  :-)

TUNER>**java -version**
openjdk version "11.0.25" 2024-10-15
OpenJDK Runtime Environment (build 11.0.25+9-post-Raspbian-1deb11u1)
OpenJDK Server VM (build 11.0.25+9-post-Raspbian-1deb11u1, mixed mode)

**My command file**
! /bin/bash
export CWD=$PROJ/Tune
export DIST="$CWD/dist"
export CLASSPATH=$DIST/lib
cd $CWD
sudo -E java -Dlog4j.ConfigurationFile=$HOME/Tune/log4j2.xml \
                -Dpi4j.library.path=$CLASSPATH -Djava.lib.path=$CLASSPATH  -jar $DIST/Tune.jar $1

**My library directory**
   All pi4j files are version 2.7
TUNER>pwd
/home/pi/NetBeansProjects/Tune/dist/lib
TUNER>dir
total 2968
-rwxrwxrwx 1 pi pi   68115 Nov 14 15:35 slf4j-api.jar
-rwxrwxrwx 1 pi pi   28051 Nov 14 15:35 pi4j-plugin-raspberrypi.jar
-rwxrwxrwx 1 pi pi   42130 Nov 14 15:35 pi4j-plugin-pigpio.jar
-rwxrwxrwx 1 pi pi  151659 Nov 14 15:35 pi4j-library-pigpio.jar
-rwxrwxrwx 1 pi pi  344441 Nov 14 15:35 pi4j-core.jar
-rwxrwxrwx 1 pi pi 1847550 Nov 14 15:35 log4j-core-2.20.0.jar
-rwxrwxrwx 1 pi pi  313296 Nov 14 15:35 log4j-api-2.20.0.jar
-rwxrwxrwx 1 pi pi   33696 Nov 14 15:35 **libpi4j-pigpio.so**
-rwxrwxrwx 1 pi pi  125099 Nov 14 15:35 Utilities.jar
-rwxrwxrwx 1 pi pi   60409 Nov 14 15:35 RaspberryPi.jar

**The exception printout**
CWD=/home/pi/NetBeansProjects/Tune
DIST=/home/pi/NetBeansProjects/Tune/dist
CLASSPATH=/home/pi/NetBeansProjects/Tune/dist/lib
/home/pi/NetBeansProjects/Tune
/home/pi/NetBeansProjects/Tune/dist
/home/pi/NetBeansProjects/Tune/dist/lib
SLF4J(W): No SLF4J providers were found.
SLF4J(W): Defaulting to no-operation (NOP) logger implementation
SLF4J(W): See https://www.slf4j.org/codes.html#noProviders for further details.
Exception in thread "main" java.lang.UnsatisfiedLinkError: Pi4J was unable to extract and load the native library [/home/pi/NetBeansProjects/Tune/dist/lib/libpi4j-pigpio.so] from the embedded resources inside this JAR [/home/pi/NetBeansProjects/Tune/dist/lib/pi4j-library-pigpio.jar]. to a temporary location on this system.  You can alternatively define the 'pi4j.library.path' system property to override this behavior and specify the library path.
 UNDERLYING EXCEPTION: [java.lang.UnsatisfiedLinkError]=/home/pi/NetBeansProjects/Tune/dist/lib/libpi4j-pigpio.so: libpigpio.so.1: cannot open shared object file: No such file or directory
	at com.pi4j.library.pigpio.util.NativeLibraryLoader.load(NativeLibraryLoader.java:145)
	at com.pi4j.library.pigpio.internal.PIGPIO.<clinit>(PIGPIO.java:74)
	at com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise(PiGpioNativeImpl.java:105)
	at com.pi4j.library.pigpio.PiGpio.initialize(PiGpio.java:159)
	at com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl.create(PiGpioI2CProviderImpl.java:77)
	at com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl.create(PiGpioI2CProviderImpl.java:46)
	at raspberrypi.I2CPi4j.<init>(I2CPi4j.java:63)
	at QWIK.ADS1015.<init>(ADS1015.java:107)
	at QWIK.ADS1015.getInstance(ADS1015.java:95)
	at HATS.StepperHat.<clinit>(StepperHat.java:45)
	at tuner.Controller.TUNE_main(Controller.java:90)
	at tuner.Controller.main(Controller.java:63)

This is why I think I got a recent version of java when I did a system upgrade two days ago after which I get the exception
TUNER>pwd
/usr/lib/jvm/java-11-openjdk-armhf/bin
TUNER>dir java
-rwxr-xr-x 1 root root 5548 **Oct 21 08:21 java**

What ever version was present before that was working. I can't remember when I upgraded last.. probably a couple months  ago

Hope this helps

FDelporte, 2024-11-16:
Just an idea... can you try sdkman and install an earlier 11 version?

RonaldTHerman, 2024-11-16:
I’m not familiar with  sdkman.. 
How do I determine what earlier versions are available?

FDelporte, 2024-11-16:
On my mac it shows:

```
$ sdk list java
...
               |     | 11.0.25.fx   | zulu    |            | 11.0.25.fx-zulu                                                                                                                         
               |     | 11.0.25      | zulu    |            | 11.0.25-zulu                                                                                                                            
               |     | 8.0.432.fx   | zulu    |            | 8.0.432.fx-zulu                                                                                                                         
               |     | 8.0.432      | zulu    |            | 8.0.432-zulu                                                                                                                            
               |     | 8.0.362      | zulu    | local only | 8.0.362-zulu
```

Another solution could be to follow these guidelines https://docs.azul.com/core/install/debian with an earlier version from https://www.azul.com/downloads/?version=java-11-lts&os=debian&package=jdk&show-old-builds=true#zulu

RonaldTHerman, 2024-11-18:
I tried the azul thing but....
  522  sudo apt install gnupg ca-certificates curl
  523  curl -s https://repos.azul.com/azul-repo.key | sudo gpg --dearmor -o /usr/share/keyrings/azul.gpg
  524  echo "deb [signed-by=/usr/share/keyrings/azul.gpg] https://repos.azul.com/zulu/deb stable main" | sudo tee /etc/apt/sources.list.d/zulu.list
  525  sudo apt update
  526  sudo apt install zulu21-jdk
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package zulu21-jdk

What am I missing?

FDelporte, 2024-11-18:
Are you running on a 32- or 64-bit system? On which type of board did you try this?

RonaldTHerman, 2024-11-18:
Pi3B+I think it will do both. Sent from my iPhone

RonaldTHerman, 2024-11-18:
It is several years old. Sent from my iPhone

FDelporte, 2024-11-18:
Sorry for the ongoing questions... ;-)
What is the output from this script?
https://github.com/FDelporte/PiJava/blob/master/scripts/pi_checks/check_raspian_version.sh

RonaldTHerman, 2024-11-18:
TUNER>./scp
Checking the Raspian version

Debian version: 11.11
Kernel version: 6.1.21-v7+

Full version info: 
Linux tuner 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

Even more version info: 
Linux version 6.1.21-v7+ ***@***.***) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP Mon Apr  3 17:20:52 BST 2023
TUNER>

RonaldTHerman, 2024-11-23:
Did the additional version info give any clues?  I'm up against a wall. What worked before doesn't anymore and every variable points to the lib and the file is in the lib.

RonaldTHerman, 2024-11-23:
Just tried pi4j-2.71-SNAPSHOT.. no change

pavly-gerges, 2025-01-14:
> UNDERLYING EXCEPTION: [java.lang.UnsatisfiedLinkError]=/home/pi/NetBeansProjects/Tune/dist/lib/libpi4j-pigpio.so: libpigpio.so.1: cannot open shared object file: No such file or directory
> at com.pi4j.library.pigpio.util.NativeLibraryLoader.load(NativeLibraryLoader.java:145)
> at com.pi4j.library.pigpio.internal.PIGPIO.(PIGPIO.java:74)
> at com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise(PiGpioNativeImpl.java:105)
> at com.pi4j.library.pigpio.PiGpio.initialize(PiGpio.java:159)
> at com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl.create(PiGpioI2CProviderImpl.java:77)
> at com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl.create(PiGpioI2CProviderImpl.java:46)
> at raspberrypi.I2CPi4j.(I2CPi4j.java:63)
> at QWIK.ADS1015.(ADS1015.java:107)
> at QWIK.ADS1015.getInstance(ADS1015.java:95)
> at HATS.StepperHat.(StepperHat.java:45)
> at tuner.Controller.TUNE_main(Controller.java:90)
> at tuner.Controller.main(Controller.java:63)

Hello, just notice that the presence of the **shared object file** in the location doesn't mean everything works fine. Please, double check if the file is corrupted, you could use the `file ./libpi4j-pigpio.so` to check it. And, maybe try to load the function tables manually using the system loader (the same one used by your JVM).

Could you try this, please? and let us know the output: 
```bash
$ readelf -h ./libpi4j-pigpio.so && ls -lhs libpi4j-pigpio.so && file ./libpi4j-pigpio.so && echo '....' && nm --demangle=java ./libpi4j-pigpio.so && echo '----' && ld -shared libpi4j-pigpio.so
```

pavly-gerges, 2025-01-14:
There are multiple things to consider, ranging from file corruptions to user permissions on this directory or particularly on the Jar file to extraction errors.

RonaldTHerman, 2025-01-14:
I made an additional comment on this recently but the comment apparently didn't make it, but I solved the problem. My memory is not totally certain but on the pi I did and upgrade of Pi4j. and noticed some things had a more recent date on them.. Apparently the "apt upgrade" I did that seemed to break things missed something or there was a recent update in the Pi4j area. All is good, thanks for your input..

closed 2025-01-14 by RonaldTHerman