Pi4J/pi4j #412: Merge of #373 prevents Building pi4j on a Pi.

taartspi, 2024-11-16:
@eitch @FDelporte    I did a fetch and rebase  and then mvn clean install -Pnative.     
 Failures: 
[ERROR]   AutoPlatformsTest.testPlatformsNotEmpty:69 expected: <false> but was: <true>
[ERROR] Errors: 
[ERROR]   I2CRawDataTest.testRawDataStream:148 » Pi4J Failed to execute action for devic...
[ERROR    

In the file DefaultContextBuilder   line 54    protected boolean autoDetectMockPlugins = !BoardInfoHelper.runningOnRaspberryPi();
      I set this to true and the build and test works.     
I did no further investigation @eitch thinking your suggestion of how to change #373 is needed.

taartspi, 2024-11-18:
@eitch   robert. Should i locate where the test environment creates the Context and use new signature ?   I think we added one that loads the needed providers. If not maybe that signature is required

eitch, 2024-11-18:
@taartspi Hi, yes this makes sense, to allow loading the mocks in test cases also on the Raspberry Pi.