Pi4J/pi4j #413: Compilation broken door checkstyle

FDelporte, 2024-11-16:
Please @dariuszzbyrad, I need your help on this one...

I tried extending checkstyle.xml with the following, but am still blocked:

```
<module name="Checker">
    <module name="TreeWalker">
        <!-- Ignore missing field/variable comments -->
        <module name="JavadocVariable">
            <property name="severity" value="ignore"/>
        </module>

        <!-- Ignore missing method comments -->
        <module name="JavadocMethod">
            <property name="allowMissingParamTags" value="true"/>
            <property name="allowMissingReturnTag" value="true"/>
        </module>

        <!-- Ignore enum constant documentation -->
        <module name="JavadocType">
            <property name="tokens" value="INTERFACE_DEF, CLASS_DEF, ANNOTATION_DEF"/>
            <!-- This excludes ENUM_DEF -->
        </module>
    </module>
</module>
```

After running `mvn clean install -Pnative -e`:

```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-checkstyle-plugin:3.5.0:check (checkstyle-check) on project pi4j-core: Failed during checkstyle execution: There are 122 errors reported by Checkstyle 9.3 with config/checkstyle/checkstyle.xml ruleset. -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-checkstyle-plugin:3.5.0:check (checkstyle-check) on project pi4j-core: Failed during checkstyle execution
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute2 (MojoExecutor.java:333)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute (MojoExecutor.java:316)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:212)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:174)
    at org.apache.maven.lifecycle.internal.MojoExecutor.access$000 (MojoExecutor.java:75)
    at org.apache.maven.lifecycle.internal.MojoExecutor$1.run (MojoExecutor.java:162)
    at org.apache.maven.plugin.DefaultMojosExecutionStrategy.execute (DefaultMojosExecutionStrategy.java:39)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:159)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:105)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:73)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:53)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:118)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:261)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:173)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:101)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:906)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:283)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:206)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:103)
    at java.lang.reflect.Method.invoke (Method.java:580)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:283)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:226)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:407)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:348)
Caused by: org.apache.maven.plugin.MojoExecutionException: Failed during checkstyle execution
    at org.apache.maven.plugins.checkstyle.CheckstyleViolationCheckMojo.execute (CheckstyleViolationCheckMojo.java:562)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:126)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute2 (MojoExecutor.java:328)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute (MojoExecutor.java:316)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:212)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:174)
    at org.apache.maven.lifecycle.internal.MojoExecutor.access$000 (MojoExecutor.java:75)
    at org.apache.maven.lifecycle.internal.MojoExecutor$1.run (MojoExecutor.java:162)
    at org.apache.maven.plugin.DefaultMojosExecutionStrategy.execute (DefaultMojosExecutionStrategy.java:39)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:159)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:105)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:73)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:53)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:118)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:261)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:173)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:101)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:906)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:283)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:206)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:103)
    at java.lang.reflect.Method.invoke (Method.java:580)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:283)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:226)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:407)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:348)
Caused by: org.apache.maven.plugins.checkstyle.exec.CheckstyleExecutorException: There are 122 errors reported by Checkstyle 9.3 with config/checkstyle/checkstyle.xml ruleset.
    at org.apache.maven.plugins.checkstyle.exec.DefaultCheckstyleExecutor.executeCheckstyle (DefaultCheckstyleExecutor.java:221)
    at org.apache.maven.plugins.checkstyle.CheckstyleViolationCheckMojo.execute (CheckstyleViolationCheckMojo.java:557)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:126)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute2 (MojoExecutor.java:328)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute (MojoExecutor.java:316)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:212)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:174)
    at org.apache.maven.lifecycle.internal.MojoExecutor.access$000 (MojoExecutor.java:75)
    at org.apache.maven.lifecycle.internal.MojoExecutor$1.run (MojoExecutor.java:162)
    at org.apache.maven.plugin.DefaultMojosExecutionStrategy.execute (DefaultMojosExecutionStrategy.java:39)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:159)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:105)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:73)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:53)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:118)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:261)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:173)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:101)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:906)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:283)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:206)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:103)
    at java.lang.reflect.Method.invoke (Method.java:580)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:283)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:226)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:407)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:348)
[ERROR] 
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :pi4j-core
```

dariuszzbyrad, 2024-11-16:
Hi @FDelporte , I'm sorry, but I'll be able to look at it in next week. The key line seems to be "Caused by: org.apache.maven.plugins.checkstyle.exec.CheckstyleExecutorException: There are 122 errors reported by Checkstyle 9.3 with config/checkstyle/checkstyle.xml ruleset." It looks like the rules defined in checkstyle were not met in 122 places.

FDelporte, 2024-11-16:
To be able to continue, I deleted the checkstyle plugin, and ignored errors in JavaDoc creation.
See https://github.com/Pi4J/pi4j-v2/pull/407/files

dariuszzbyrad, 2024-11-20:
@FDelporte please assign me to it. I'll fix it.

dariuszzbyrad, 2024-11-20:
Fixed in the [https://github.com/Pi4J/pi4j-v2/pull/417](https://github.com/Pi4J/pi4j-v2/pull/417)

FDelporte, 2024-11-20:
Thanks for the merge request @dariuszzbyrad.

`mvn clean package` now succeeds. But I still see some errors. Should these be further investigated?

For instance: 

```
[ERROR] MavenReportException: Error while generating Javadoc: 
Exit code: 2
error: Illegal package name: "ALL-MODULE-PATH"
1 error
Command line was: /Users/frank/.sdkman/candidates/java/22.0.2.fx-zulu/zulu-22.jdk/Contents/Home/bin/javadoc ALL-MODULE-PATH @options @packages @argfile

Refer to the generated Javadoc files in '/Users/frank/Documents/GitHub/pi4j-v2/pi4j-test/target/apidocs' dir.

org.apache.maven.reporting.MavenReportException: 
Exit code: 2
error: Illegal package name: "ALL-MODULE-PATH"
1 error
Command line was: /Users/frank/.sdkman/candidates/java/22.0.2.fx-zulu/zulu-22.jdk/Contents/Home/bin/javadoc ALL-MODULE-PATH @options @packages @argfile

Refer to the generated Javadoc files in '/Users/frank/Documents/GitHub/pi4j-v2/pi4j-test/target/apidocs' dir.

```

dariuszzbyrad, 2024-11-20:
For what exact command are you getting the above error? Please also provide result for 
```bash
mvn --version
```

From my perspective:
```bash
java --version               
openjdk 21.0.2 2024-01-16 LTS
OpenJDK Runtime Environment Temurin-21.0.2+13 (build 21.0.2+13-LTS)
OpenJDK 64-Bit Server VM Temurin-21.0.2+13 (build 21.0.2+13-LTS, mixed mode)
```

The `mvn clean install -Pnative -e` and `mvn clean package` returns `BUILD SUCCESS`

When I switch to your java version:
```bash
java --version                 
openjdk 22.0.2 2024-07-16
OpenJDK Runtime Environment Zulu22.32+15-CA (build 22.0.2+9)
OpenJDK 64-Bit Server VM Zulu22.32+15-CA (build 22.0.2+9, mixed mode, sharing)
```

The `mvn clean install -Pnative -e` and `mvn clean package` also returns `BUILD SUCCESS`

FDelporte, 2024-11-20:
@dariuszzbyrad the build succeeds, these errors are shown in the build logs. Command is `mvn clean package`.

dariuszzbyrad, 2024-11-21:
Ok, now I know a bit more. Problems are two:
**1. Javadoc Configuration Issue** 
**Problem**:

The Javadoc configuration in the `pom.xml `contained unnecessary and incorrect options:

```xml
<additionalJOption>--add-modules</additionalJOption>
<additionalJOption>ALL-MODULE-PATH</additionalJOption>
```

**Why it's problematic:**
These options are only needed if the project explicitly requires additional modules, and `ALL-MODULE-PATH` isn't always appropriate.
If the project is modular, `--module-path` should be configured correctly to point to the module path.

**Solution:**

Remove the problematic lines and adjust the configuration to focus on your project's specific needs. For modular projects, ensure the correct `--module-path` is provided, or omit these options entirely if they are unnecessary.

**2. Warning:** `no description for @param`
**Problem:**

Warnings were generated for missing descriptions of `@param` tags in Javadoc, requiring either extensive manual fixes or suppression.

Solution: 
Add description for all `@param` tag or suppress these warnings using:
`<additionalOptions>-Xdoclint:none</additionalOptions>`

### Final Javadoc Configuration
```xml
<plugin>
                    <groupId>org.apache.maven.plugins</groupId>
                    <artifactId>maven-javadoc-plugin</artifactId>
                    <version>${maven-javadoc-plugin.version}</version>
                    <executions>
                        <execution>
                            <id>attach-javadocs</id>
                            <goals>
                                <goal>jar</goal>
                            </goals>
                        </execution>
                    </executions>
                    <configuration>
                        <source>${java.version}</source>
                        <detectJavaApiLink>false</detectJavaApiLink>
                        <doclint>none</doclint>
                        <!-- Optional: skip errors -->
                        <failOnError>false</failOnError>
                        <additionalOptions>-Xdoclint:none</additionalOptions>
                    </configuration>
                </plugin>
```
or by patch:
```
Index: pom.xml
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/pom.xml b/pom.xml
--- a/pom.xml	(revision f1ea6ce777729e6db8610a5688a2ac22fe696820)
+++ b/pom.xml	(date 1732176217052)
@@ -506,11 +506,9 @@
                         <source>${java.version}</source>
                         <detectJavaApiLink>false</detectJavaApiLink>
                         <doclint>none</doclint>
-                        <!-- Add module options -->
-                        <additionalJOption>--add-modules</additionalJOption>
-                        <additionalJOption>ALL-MODULE-PATH</additionalJOption>
                         <!-- Optional: skip errors -->
                         <failOnError>false</failOnError>
+                        <additionalOptions>-Xdoclint:none</additionalOptions>
                     </configuration>
                 </plugin>
``` 

### PR
[https://github.com/Pi4J/pi4j-v2/pull/418](https://github.com/Pi4J/pi4j-v2/pull/418)