Pi4J/pi4j #421: Unable to detect board type

byRizon, 2024-11-22:
BoardInfoHelper.class is unable to detect my RaspberryPi 5 board type due to error:
`Could not execute cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}' to detect the board model: IOException: Cannot run program "sh": error=13, Permission denied`

The .jar file is executed by the root user.

FDelporte, 2024-11-23:
auch, that is strange

do these commands work as root user in the terminal?

```
cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'
cat /proc/device-tree/model
```

dariuszzbyrad, 2024-11-23:
Could you please provide the result of the `ls -l /bin/sh` command? This will help us improve the functionality.

byRizon, 2024-11-23:
yes those commands work as a root user

[image]

[image]

FDelporte, 2024-11-28:
@byRizon can you please try version 2.7.1-SNAPSHOT of Pi4J? @dariuszzbyrad improved the board detection process to prevent this error from happening.

You will probable need to add this to your pom:

```
<repositories>
    <repository>
        <id>oss-snapshots-repo</id>
        <name>Sonatype OSS Maven Repository</name>
        <url>https://oss.sonatype.org/content/groups/public</url>
        <releases>
            <enabled>false</enabled>
        </releases>
        <snapshots>
            <enabled>true</enabled>
        </snapshots>
    </repository>
</repositories>
```

byRizon, 2024-11-28:
I've tested the 2.7.1-SNAPSHOT of Pi4J and some errors still occur. After some further testing and tinkering around i solved the issue.
Pi4J is able to read the board code, memory and the cpu revision with the 2.7.1-SNAPSHOT.
For some reason the root user didn't have access to the "jspawnhelper" file in the Java home. Thats why Pi4J was still unable to perform any commands via the ProcessBuilder in the CommandExecutor.

`chmod +x JAVA_HOME/lib/jspawnhelper` -> solved the issue for me.

dariuszzbyrad, 2024-11-29:
Is this **exactly** the same error?
`
Could not execute cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}' to detect the board model: IOException: Cannot run program "sh": error=13, Permission denied`

If so, it seems you're using an older version (prior to my fix). In the updated version, we no longer execute the above command. Instead, we directly read the `/proc/cpuinfo` file and perform the same transformation in Java.

byRizon, 2024-11-30:
As is said before `Pi4J is able to read the board code, memory and the cpu revision with the 2.7.1-SNAPSHOT.`
ツ

closed 2024-12-13 by byRizon