Pi4J/pi4j #423: GPIOD on Ubuntu requires sudo

thetechnician94, 2024-11-25:
When starting Pi4J with GpioD on Ubuntu, you get
'com.pi4j.library.gpiod.internal.GpioDException: gpiod_chip_iter_new failed! (Permission denied)
at com.pi4j.library.gpiod.internal.GpioD.chipIterNew(GpioD.java:344
..
..'

Running as root/sudo works as expected. 

Adding the user to the 'dialout' group as described here: https://askubuntu.com/questions/1230947/gpio-for-raspberry-pi-gpio-group
does not seem to work either.

dariuszzbyrad, 2024-12-19:
Please double-check that your user has been successfully added to the correct group. You can verify this by running the command:

`groups <your_username>`

Make sure the gpio group (or dialout, depending on your setup) appears in the output. If it’s not listed, you may need to log out and log back in, or restart your system for the group membership to take effect."

thetechnician94, 2024-12-19:
[image]
Everything appears correct