Pi4J/pi4j #431: PIGPIO ERROR: PI_INIT_FAILED; pigpio initialization failed

Diabolo-kun, 2024-12-18:
﻿
Hi, i compiled my project in a .jar. Myproject consist in a raspberry pi conected at a M5Stack by UART, when i excute the jar in the rasp's terminal show this:
```
@raspberrypi:~/Desktop $ java -jar raspby.jar [main] INFO com.pi4j.Pi43 New auto context
[main] INFO com.pi4j.Pi43 New context builder [main] INFO con.pi4j.boardinfo.util.BoardInfoHelper [main] INFO com.pi4j.boardinfo.util.BoardInfoHelper [main] INFO com.pi4j.boardinfo.util.BoardInfoHelper [main] INFO com.pi4j.context.impl.DefaultContext [main] INFO com.pi4j.context.impl.DefaultContext [main] INFO com.pi4j.context.impl.DefaultContext [main] INFO com.pi4j.runtime.impl.DefaultRuntime [main] INFO com.pi4j.runtime.impl.DefaultRuntime
Detected OS: Name: Linux, version: 6.6.62+rpt-rpi-v7, architecture: arm
Detected Java: Version: 17.0.13, runtime: 17.0.13+11-Raspbian-2deb12u1rpt1, vendor: Raspbian, vendor version: null Detected board type MODEL_3_8 by code: a02082
Detected board model: Raspberry Pi 3 Model B
Running on: Name: Linux, version: 6.6.62+rpt-rpi-v7, architecture: arm
With Java version: Version: 17.0.13, runtime: 17.0.13-11-Raspbian-2deb12u1rpt1, vendor: Raspbian, vendor version: null Initializing Pi43 context/runtime...
Pi43 context/runtime successfully initialized.
[main] WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed Error al inicializar la conexión serial: PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed com.pi4j.library.pigpio.PiepioException: PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
at com.pi4j.library.pigpio.impl.PiopioBase.validateResult(Pi6pioBase.java:265)
at com.pi4j.library.pigpio.impl.PiopioBase.validateResult(PiopioBase.java:251)
at com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise (PiGpioNativeImpl.java:110) at com.pi4j.library.pigpio.Pipio.initialize(Piopio.java:159)
at com.pi4j.plugin.pigpio.provider.serial.
PicpioSerialProviderImpl.create(PiGpioSerialProviderImpl.java:71)
at com.pi4j.plugin.pigpio.provider.serial.PiGpioSerialProviderImpl.create(PiGpioSerialProviderImpl.java:42) at com.pi4j.context.Context.create(Context.java:344)
at
at
com.pi4j.internal. I0Creator.create(10Creator.java:58) com.pi4j.internal. I0Creator.create(IDCreator.java:146)
at com.example.raspby.M5StackHandler.<init>(M5StackHandler.java:20)
at com.example.raspby.MainProgram.main(MainProgram.java:25)
```
this is the class where i use everything relacioned with pi4j and the UART communication:
package com.example.raspby;
```

import com.pi4j.io.serial.*;
import com.pi4j.context.Context;
import com.pi4j.Pi4J;

public class M5StackHandler {
    private Serial serial;
    private boolean continueReading = true;

    public M5StackHandler(String portName, int baudRate) {
        int maxRetries = 5;  // Número máximo de reintentos
        int attempt = 0;  // Intento actual
        boolean connected = false;

        while (attempt < maxRetries && !connected) {
            try {
                // Configurar y abrir el puerto serial con Pi4J
                Context pi4j = Pi4J.newAutoContext();
                serial = pi4j.create(Serial.newConfigBuilder(pi4j)
                        .use_115200_N81()
                        .dataBits_8()
                        .parity(Parity.NONE)
                        .stopBits(StopBits._1)
                        .flowControl(FlowControl.NONE)
                        .id("my-serial")
                        .device(portName) // Usar el puerto proporcionado, por ejemplo "/dev/ttyS0"
                        .provider("pigpio-serial")  // Especificar el proveedor correcto (verifica que esté disponible)
                        .build());

                serial.open();
                System.out.println("Conexión serial con M5Stack establecida.");
                connected = true;  // Si la conexión se establece, salimos del bucle

            } catch (Exception e) {
                attempt++;
                System.out.println("Error al inicializar la conexión serial: " + e.getMessage());
                e.printStackTrace();
                if (attempt < maxRetries) {
                    System.out.println("Reintentando... (" + attempt + "/" + maxRetries + ")");
                    try {
                        Thread.sleep(2000);  // Esperar 2 segundos antes de reintentar
                    } catch (InterruptedException ie) {
                        Thread.currentThread().interrupt();  // Restablecer el estado de interrupción
                    }
                } else {
                    System.out.println("Número máximo de reintentos alcanzado. No se pudo establecer la conexión.");
                }
            }
        }
    }

    // Método para enviar un mensaje al M5Stack
    public void sendMessage(String message) {
        System.out.println("Enviando mensaje: " + message); // Verifica si el mensaje es correcto

        if (message == null || message.trim().isEmpty()) {
            System.out.println("El mensaje está vacío o es nulo.");
            return;
        }

        try {
            if (serial.isOpen()) {
                serial.write(message + "\n");
                System.out.println("Mensaje enviado al M5Stack: " + message);
            } else {
                System.out.println("El puerto serial no está abierto.");
            }
        } catch (Exception e) {
            System.out.println("Error al enviar mensaje: " + e.getClass().getName() + " - " + e.getMessage());
        }
    }

    // Cerrar la conexión serial
    public void close() {
        try {
            if (serial.isOpen()) {
                serial.close();
                System.out.println("Conexión serial cerrada.");
            }
        } catch (Exception e) {
            System.out.println("Error al cerrar la conexión serial: " + e.getMessage());
        }
    }

    // Leer mensaje del puerto serial
    public String readMessage() {
        StringBuilder lineBuilder = new StringBuilder();
        try {
            int available = serial.available();
            if (available > 0) {
                for (int i = 0; i < available; i++) {
                    byte b = serial.readByte();
                    if (b < 32) { // Fin de mensaje (por ejemplo, si se recibe un carácter de control)
                        if (lineBuilder.length() > 0) {
                            String line = lineBuilder.toString();
                            System.out.println("Recibido de M5: " + line);
                            return line;
                        }
                    } else {
                        lineBuilder.append((char) b);
                    }
                }
            }
            return null;
        } catch (Exception e) {
            System.out.println("Error leyendo datos del puerto serial: " + e.getMessage());
        }
        return null;
    }
}

```
I would greatly appreciate for any help to solve this problem.

Diabolo-kun, 2024-12-18:
The package pigpio is already installed, and the service are enable also.
I working with a Raspberry 3

dariuszzbyrad, 2024-12-18:
Please check the following:

1. Is pigpio installed?
2. Is the pigpio daemon started?
3. Are you running the application with sudo, or is your user added to the gpio group?
4. Verify the UART connection works and is pointing to the correct device. Check the output of:

```
ls -l /dev/serial*
ls -l /dev/ttyS*
```

5. Did you test the UART manually?
Use a tool like minicom to manually test the UART communication.