Pi4J/pi4j #445: Exception using PWM

DieterHolz, 2025-02-27:
I tried to get the PWM example "BuzzerApp" from "pi4j-example-components" working according to blogpost https://www.pi4j.com/blog/2024/20240423_pwm_rpi5/ 

Code is on branch "Pi5"

As far as I remember it worked sometimes, but now it throws this exception when calling pwm.off().

The rest seems to work (the buzzer makes the expected noise)

[main] ERROR com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm - Invalid argument
java.io.IOException: Invalid argument
	at java.base/sun.nio.ch.UnixFileDispatcherImpl.write0(Native Method)
	at java.base/sun.nio.ch.UnixFileDispatcherImpl.write(UnixFileDispatcherImpl.java:65)
	at java.base/sun.nio.ch.IOUtil.writeFromNativeBuffer(IOUtil.java:137)
	at java.base/sun.nio.ch.IOUtil.write(IOUtil.java:102)
	at java.base/sun.nio.ch.IOUtil.write(IOUtil.java:72)
	at java.base/sun.nio.ch.FileChannelImpl.write(FileChannelImpl.java:300)
	at java.base/sun.nio.ch.ChannelOutputStream.writeFully(ChannelOutputStream.java:68)
	at java.base/sun.nio.ch.ChannelOutputStream.write(ChannelOutputStream.java:105)
	at java.base/java.nio.file.Files.write(Files.java:3510)
	at java.base/java.nio.file.Files.writeString(Files.java:3727)
	at java.base/java.nio.file.Files.writeString(Files.java:3667)
	at com.pi4j.plugin.linuxfs.internal.LinuxPwm.setEnabled(LinuxPwm.java:244)
	at com.pi4j.plugin.linuxfs.internal.LinuxPwm.disable(LinuxPwm.java:221)
	at com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.off(LinuxFsPwm.java:161)
	at com.pi4j.catalog.components.Buzzer.off(Buzzer.java:77)
	at com.pi4j.catalog.components.Buzzer.<init>(Buzzer.java:44)
	at com.pi4j.catalog.applications.BuzzerApp.main(BuzzerApp.java:56)
Exception in thread “main” [com.pi4j.io](http://com.pi4j.io/).exception.IOException: Unable to disable (OFF) PWM [2] @ </sys/class/pwm/pwmchip2/pwm2>; Invalid argument
	at com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.off(LinuxFsPwm.java:168)
	at com.pi4j.catalog.components.Buzzer.off(Buzzer.java:77)
	at com.pi4j.catalog.components.Buzzer.<init>(Buzzer.java:44)
	at com.pi4j.catalog.applications.BuzzerApp.main(BuzzerApp.java:56)

pi4j.com
[PWM Hardware Support on RPi5 - Pi4J](https://www.pi4j.com/blog/2024/20240423_pwm_rpi5/)
Java IO Library for the Raspberry Pi
[5:11](https://pi4j.slack.com/archives/D01HRQLHPL7/p1740154263106769)

taartspi, 2025-03-01:
@DieterHolz     I set up my Pi5 using your Pi5 branch.    I can go to the Application package BuzzerApp and run your app.  As I don't have a buzzer just watching a connected LED and I see the program functions.   This is working ok.

Is there a different manner I should us to run the application ?
Or is it just repeat execution and see if I hit the exception ?
thanks

DieterHolz, 2025-03-01:
@taartspi 
It must be a setup problem of the Pi, then. The exception occurs every single time if you call pwm.off()

I've just added 

dtoverlay=pwm-2chan

to the config.txt.

Is there anything else to do?

taartspi, 2025-03-03:
pi@rasp5Dev:/sys/class/pwm/pwmchip2 $ ls
device  export  npwm  power  subsystem  uevent  unexport
pi@rasp5Dev:/sys/class/pwm/pwmchip2 $ ls
device  export  npwm  power  **pwm2**  subsystem  uevent  unexport
pi@rasp5Dev:/sys/class/pwm/pwmchip2 $ 

Of significance, the channel has been enabled,  so pwm.off() can function.   

This is my bug. If the first thing attempted is off, I do not enable channel.   So, while I fix this, If your team wants to make progress on this test, in the Buzzer class, after creating the configBuilder, comment out the off().     

I am busy much of Monday and Tuesday and may find it difficult to fix this that soon.   Tom...    

So the other night when I could not duplicate you error, I first used a pwm tests of my own and it enabled the channel, so than your tests would function for me.

DieterHolz, 2025-03-03:
Take your time. 

Avoiding an initial  pwm.off() is an easy, good enough workaround.

FDelporte, 2025-03-03:
Another exception, not sure if related. 
Using this example: https://github.com/Pi4J/pi4j-jbang/blob/main/Pi4JPWMLed.java
It works, but throws an exception in the beginning:

```
Starting PWM output example...
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected OS: Name: Linux, version: 6.6.74+rpt-rpi-v8, architecture: aarch64
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected Java: Version: 21.0.5, runtime: 21.0.5+11-LTS, vendor: Azul Systems, Inc., vendor version: Zulu21.38+21-CA
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected board type MODEL_4_B by code: c03112
[main] INFO com.pi4j.context.impl.DefaultContext - Detected board model: Raspberry Pi 4 Model B
[main] INFO com.pi4j.context.impl.DefaultContext - Running on: Name: Linux, version: 6.6.74+rpt-rpi-v8, architecture: aarch64
[main] INFO com.pi4j.context.impl.DefaultContext - With Java version: Version: 21.0.5, runtime: 21.0.5+11-LTS, vendor: Azul Systems, Inc., vendor version: Zulu21.38+21-CA
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Initializing Pi4J context/runtime...
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider I2C RaspberryPi I2C Provider with priority 0 with provider PiGpio I2C Provider with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider SPI RaspberryPi SPI Provider with priority 0 with provider PiGpio SPI Provider with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider DIGITAL_OUTPUT RaspberryPi Digital Output (GPIO) Provider with priority 0 with provider PiGpio Digital Output (GPIO) Provider with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider SERIAL RaspberryPi Serial Provider with priority 0 with provider PiGpio Serial Provider with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider DIGITAL_INPUT RaspberryPi Digital Input (GPIO) Provider with priority 0 with provider PiGpio Digital Input (GPIO) Provider with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider PWM RaspberryPi PWM Provider with priority 0 with provider PiGpio PWM Provider with higher priority 100
[main] INFO com.pi4j.platform.impl.DefaultRuntimePlatforms - adding platform to managed platform map [id=raspberrypi; name=RaspberryPi Platform; priority=5; class=com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform]
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Pi4J context/runtime successfully initialized.
[main] ERROR com.pi4j.library.pigpio.impl.PiGpioNativeImpl - /sys/module/spidev/parameters/bufsiz (No such file or directory)
java.io.FileNotFoundException: /sys/module/spidev/parameters/bufsiz (No such file or directory)
        at java.base/java.io.FileInputStream.open0(Native Method)
        at java.base/java.io.FileInputStream.open(FileInputStream.java:213)
        at java.base/java.io.FileInputStream.<init>(FileInputStream.java:152)
        at java.base/java.util.Scanner.<init>(Scanner.java:645)
        at com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise(PiGpioNativeImpl.java:113)
        at com.pi4j.library.pigpio.PiGpio.initialize(PiGpio.java:159)
        at com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProviderImpl.create(PiGpioPwmProviderImpl.java:73)
        at com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProviderImpl.create(PiGpioPwmProviderImpl.java:44)
        at com.pi4j.context.Context.create(Context.java:363)
        at com.pi4j.internal.IOCreator.create(IOCreator.java:58)
        at com.pi4j.internal.IOCreator.create(IOCreator.java:116)
        at Pi4JPWMLed.main(Pi4JPWMLed.java:49)
PWM duty cycle / frequency is: 50.0/800
PWM duty cycle / frequency is: 5.0/1500
PWM duty cycle / frequency is: 10.0/1500
PWM duty cycle / frequency is: 15.0/1500
PWM duty cycle / frequency is: 20.0/1500
PWM duty cycle / frequency is: 25.0/1500
PWM duty cycle / frequency is: 30.0/1500
PWM duty cycle / frequency is: 35.0/1500
PWM duty cycle / frequency is: 40.0/1500
PWM duty cycle / frequency is: 45.0/1500
PWM duty cycle / frequency is: 50.0/1500
PWM duty cycle / frequency is: 55.0/1500
PWM duty cycle / frequency is: 60.0/1500
PWM duty cycle / frequency is: 65.0/1500
PWM duty cycle / frequency is: 70.0/1500
PWM duty cycle / frequency is: 75.0/1500
PWM duty cycle / frequency is: 80.0/1500
PWM duty cycle / frequency is: 85.0/1500
PWM duty cycle / frequency is: 90.0/1500
PWM duty cycle / frequency is: 95.0/1500
PWM duty cycle / frequency is: 100.0/1500
```

taartspi, 2025-03-03:
The bug Dieter hit is specific to my provider so this case in pigpio needs separate attention

FDelporte, 2025-03-03:
@taartspi thanks, deleted, will make separate ticket

taartspi, 2025-03-13:
@DieterHolz   Sir, I was coding up the fix for the bug you hit and was concerned that other test paths were not executed.  The bug you hit is one item, but, had an application called the shutdown interface what next actions from that application would you expect to function cleanly ?   thanks Tom

DieterHolz, 2025-03-13:
@taartspi 
Hi Tom, 
I'm sorry, but I don't understand your question. 

Is it what actions should be available after a Pi4J-shutdown? As far as I remember I don't call any actions after a shutdown.

regards,
 Dieter

taartspi, 2025-03-13:
Morning.  I think you answered me. I will assume if an application calls
Shutdown it will attempt no more pwm interface calls.

FDelporte, 2025-03-17:
Thanks for reporting @DieterHolz. Got fixed by @taartspi in #452

closed 2025-03-17 by FDelporte