Pi4J/pi4j #447: LinuxFsSpi throws exception for non-null flags, but config builder doesn't support null flags

mpilone, 2025-03-02:
With v2.8.0 the LinuxFs provider added SPI support. I attempted to switch my local code over to use the new support, but I ran into a simple issue. My original instantiation of the SPI looks like:

```
var spiConfig = Spi.newConfigBuilder(pi4jContext)
                .address(0)
                .baud(24000000)
                .build();
var spi = pi4jContext.spi().create(spiConfig);
```
This results in the exception:
```
Caused by: com.pi4j.io.exception.IOException: Unsupported SPI Pi5 parameter flags
	at com.pi4j.plugin.linuxfs.provider.spi.LinuxFsSpi.open(LinuxFsSpi.java:172) ~[pi4j-plugin-linuxfs-2.8.0.jar:?]
	at com.pi4j.plugin.linuxfs.provider.spi.LinuxFsSpiProviderImpl.create(LinuxFsSpiProviderImpl.java:64) ~[pi4j-plugin-linuxfs-2.8.0.jar:?]
```

It looks like LinuxFsSpi requires a null flags on the config because they are not supported. I then modified my SPI creation to:
```
var spiConfig = Spi.newConfigBuilder(pi4jContext)
                .address(0)
                .baud(24000000)
                .flags(null)
                .build();
        var spi = pi4jContext.spi().create(spiConfig);
```
This results in the exception:
```
Caused by: java.lang.NullPointerException: Cannot invoke "java.lang.Long.toString()" because "flags" is null
	at com.pi4j.io.spi.impl.DefaultSpiConfigBuilder.flags(DefaultSpiConfigBuilder.java:82) ~[pi4j-core-2.7.0.jar:?]
```

The config builder attempts to put the flags into a map by calling toString on the Long without a null check. This happens elsewhere in the config builder as well, such as with `rate` and `channel`. The default `flags` value is 0L in `DefaultSpiConfig`, not null. That makes me wonder why the config builder uses object types for these primitives when null is clearly not supported or intended.

There are two possible fixes:
1. Support null values for the primitives in the DefaultSpiConfigBuilder and DefaultSpiConfig, OR
2. Update LinuxFsSpi to check for `flags != 0`, not `flags != null` and accept that null can't be used in the config even though the object type primitives are used on the builder methods.

-mike

taartspi, 2025-03-03:
@FDelporte @eitch      I would like you both to comment on this entry.   The configBuilder NOT working properly with 'null' parameter values is not unique to the SPI addition I did.   Modify the address parameter value to null when creating an I2C device and its create will fail attempting toString on that null.     
I don't think this is wrong,  maybe not idea,  but at least I don't know of any slick way to change the behavior.

eitch, 2025-03-07:
Basically it is sad, that the config stores everything as Strings. It could just as well have been Object, then all these parsings wouldn't be necessary. I think we just need to fix the code, so that this works:

```java
var spiConfig = Spi.newConfigBuilder(pi4jContext)
                .address(0)
                .baud(24000000)
                .build();
var spi = pi4jContext.spi().create(spiConfig);
```

But what i don't understand is where the flags get set to non-null, as not setting them, should keep it null.

taartspi, 2025-03-07:
I believe what u typed does work.  It is when the user includes flags(null) it fails   My answer was flags and other values fail when set to the null and is not a bug. How it works

eitch, 2025-03-07:
@taartspi From what @mpilone wrote, it doesn't work as it throws an exception:

```java
Caused by: com.pi4j.io.exception.IOException: Unsupported SPI Pi5 parameter flags
	at com.pi4j.plugin.linuxfs.provider.spi.LinuxFsSpi.open(LinuxFsSpi.java:172) ~[pi4j-plugin-linuxfs-2.8.0.jar:?]
	at com.pi4j.plugin.linuxfs.provider.spi.LinuxFsSpiProviderImpl.create(LinuxFsSpiProviderImpl.java:64) ~[pi4j-plugin-linuxfs-2.8.0.jar:?]
```

taartspi, 2025-03-07:
Using 2.8.0 I ran three tests, one with the expected builder parms,  second passing a valid flags parm, third flags parm value null.  The first should work, second Pi4j throwns an exception third the OS throws.

@mpilone @eitch     Mike do you agree in these three outcomes, that they are what occurs  ?

[image]

This first one is the normal use of the builder

[image]

This second one passes a valid flags value, and an exception is thrown stating this is not supported

[image]

This third one, I pass a null value for the flags, this is an exception that the OS creates.  I think it not a problem as the null should not be passed.

mpilone, 2025-03-07:
Ok, apologies. I think I had a mix of 2.7 and 2.8 on my class path when I ran into this issue. In 2.7, the DefaultSpiConfig set the flags to 0L:

```
// load optional FLAGS BITS from properties
        if(properties.containsKey(FLAGS_KEY)){
            this.flags = StringUtil.parseLong(properties.get(FLAGS_KEY), null);
        } else {
            this.flags = 0L; // default flags (0)
        }
```

but in 2.8 that was changed to set the flags to null:

```
// load optional FLAGS BITS from properties
        if(properties.containsKey(FLAGS_KEY)){
            this.flags = StringUtil.parseLong(properties.get(FLAGS_KEY), null);
        } else {
            this.flags = null; // set null, same as parseLong would
        }
```

which matches what the `LinuxFsSpi` is expecting.

I must have updated the version in my pom and just done a build, not a clean build, which causes it to load the DefaultSpiConfig from 2.7 and trigger the exception. Sorry for wasting your time.

closed 2025-03-07 by mpilone

taartspi, 2025-03-07:
Hey @mpilone thanks for the update. Tom