Pi4J/pi4j #454: Implement new FFM API approach

DigitalSmile, 2025-03-26:
Goal: create a plugin, that can talk to hardware using [FFM API](https://openjdk.org/jeps/454). It was introduced in JDK22 and is going to replace old JNI in future releases of JDK. For now it is marked for [restricted use](https://openjdk.org/jeps/472) since next JDK25.

Todo list:

- [ ] Write some code, that works with FFM API objects (layouts, arenas, function calls, etc.)
- [ ] Implement all basic `Provider` interfaces (gpio, i2c, spi, pwm)
- [ ] Write JMH tests to measure performance
- [ ] Cover methods with tests and javadocs

AlexR1973, 2025-03-27:
@DigitalSmile Please add implementation of 1Wire support also.

p.s. Is it possible for you to add support of 1Wire in your gpio library?

DigitalSmile, 2025-03-28:
> @DigitalSmile Please add implementation of 1Wire support also.
p.s. Is it possible for you to add support of 1Wire in your gpio library?

I didn't really work with 1-wire devices. Isn't it the protocol that can work above simple GPIO Pin?
Guess it is more high level application protocol so I don't know if it is good idea to add it into pi4j. Maybe @FDelporte can step in to this discussion?

FDelporte, 2025-03-28:
About 1Wire support: there is work in progress by @dariuszzbyrad 
See https://github.com/Pi4J/pi4j/pull/435

eitch, 2025-03-28:
Additionally the FFM-API approach isn't required, as one can directly read the kernel's relevant sys files. E.g. reading a temperature device:

```java
String devicePath = "/sys/bus/w1/devices/28-0316a2791cff/w1_slave";
        try (BufferedReader br = new BufferedReader(new FileReader(devicePath))) {
            String line1 = br.readLine();  // e.g. "00 00 00 00 00 ... : crc=00 YES"
            String line2 = br.readLine();  // e.g. "00 00 00 00 00 t=12345"

            // Typically you look for "t=NNNNN" to get the temperature
            // Here, parse the integer after "t="
            String tempStr = line2.split("t=")[1];
            double tempC = Double.parseDouble(tempStr) / 1000.0;
            System.out.println("Temperature: " + tempC + "°C");
        } catch (IOException e) {
            e.printStackTrace();
        }
```

This is what @dariuszzbyrad  has implemented.

AlexR1973, 2025-03-28:
Hello guys!
I know "/sys/bus/w1/devices/28-XXXX/w1_slave" location and currently read data from it in my project. But this data provided by OS driver. As far as I know there is another way to read data from the bus directly. Don't ask me how - I have a very basic knowledge about all these protocols and hardware implementations.
Disadvantage of reading data from /sys/bus/w1/devices/28-XXXX/w1_slave location, that it refreshed every 4-10 seconds (depends on the OS settings and implementation). If you need a more accurate value - read it directly from the bus.
p.s. If you interested my project is a home brewing system controller. I have 2 DS18B20, 3 relays for solenoid valves and a BPM180 pressure monitor to adjust temperature according to outside pressure. In a test environment on Java 17, Raspberry Pi ver2 rev B it can work for hours - no issues at all. But Orane Pi zero 3 is a completely different story - PITA in any step :(
Friend of mine asked me to do the same system for him and gave me his Orange Pi. I didn't know that 2 systems have such different guts inside them.

FDelporte, 2025-03-28:
> home brewing system controller

That calls for a new page in https://www.pi4j.com/featured-projects/! ;-)

eitch, 2025-03-29:
@AlexR1973 from what i can see, there is no other way to read 1-wire devices in the kernel. At least nothing that i can find. if it takes 4-10 seconds, then perhaps its an implementation detail? Do you have any example code?

AlexR1973, 2025-03-29:
@eitch , I have found the next article about working with DS18B20 over 1-wire interface. Please take a look at it - maybe it reveals the details how to get data directly. It is in Russian, but Chrome/MS Edge will translate it to English easily.

https://micro-pi.ru/%d0%bf%d0%be%d0%b4%d0%ba%d0%bb%d1%8e%d1%87%d0%b5%d0%bd%d0%b8%d0%b5-ds18b20-%d0%ba-orange-pi-bpi-rpi/

AlexR1973, 2025-03-29:
@eitch , 
> if it takes 4-10 seconds
4 seconds update time is not a showstopper in my case, even 10 secs is acceptable. But better to have the data when you need it, not when OS decides to update it for you.

AlexR1973, 2025-03-29:
I spent 8 hours yesterday to add support of 1-wire to Orange Pi Zero 3 and Raspi OS. If you interested, I had to do next:
=== add support of 1-wire on pin PC10 pin ===

Create file sun50i-h616-w1-gpio.dts and put inside:
==========CUT BELOW===========
/dts-v1/;
/plugin/;
/ {
    model = "OrangePi Zero3";
    compatible = "xunlong,orangepi-zero3", "allwinner,sun50i-h618";
    fragment@0 {
            target-path = "/";
            __overlay__ {
                    w1: onewire@0 {
                            compatible = "w1-gpio";
                            pinctrl-names = "default";
                            gpios = <&pio 2 10 0>;
                            status = "okay";
                    };
            };
    };
};
==========CUT ABOWE===========
Compile it with command
dtc -O dtb -o sun50i-h616-w1-gpio.dtbo sun50i-h616-w1-gpio.dts
Copy sun50i-h616-w1-gpio.dtbo to /boot/dtb/allwinner/overlay directory

Add to /boot/orangepiEnv.txt:
overlays=w1-gpio //add to the end of line
param_w1_pin=pc10
param_w1_pin_int_pullup=1

edit /etc/modules-load.d/modules.conf. Add 2 lines:
w1-gpio
w1-therm

Execute after editing config files:
sudo modprobe w1-gpio
sudo modprobe w1-therm

AlexR1973, 2025-03-29:
@DigitalSmile , BTW could you open a discussion section in your "gpio" project? I have a question about your library and I2C protocol, but don't know how to contact you.

I have very simple code trying to get pressure from BMP180.
"i2cdetect -y 3" command shows that BMP180 is up and running on I2C bus №3 with 0x77 ID:
_
70: -- -- -- -- -- -- -- 77
_

Want to get pressure data from the sensor. My code:

`
            var i2c = GPIOBoard.ofI2C(3);
            var scanned = i2c.scan();
            var status = scanned.get(0x77);
            if (status.equals(I2CStatus.AVAILABLE))
            {
                i2c.selectAddress(0x77);
                var read = i2c.read(0x77);
                System.out.println("Data="+read);
                //do stuff with data read
            }
`

But all the time I got the next exception:
_
[main] INFO org.digitalsmile.gpio.i2c.I2CBus - /dev/i2c-3 - setting up I2CBus...
Exception in thread "main" java.lang.NoClassDefFoundError: org/digitalsmile/gpio/core/ioctl/Command
	at org.digitalsmile.gpio.i2c.I2CBus.<init>(I2CBus.java:63)
	at org.digitalsmile.gpio.GPIOBoard.ofI2C(GPIOBoard.java:138)
	at org.digitalsmile.gpio.GPIOBoard.ofI2C(GPIOBoard.java:149)
	at com.BoosMaster.Main.main(Main.java:17)
Caused by: java.lang.ClassNotFoundException: org.digitalsmile.gpio.core.ioctl.Command
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:580)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:490)
	... 4 more
_
org.digitalsmile.gpio.core.ioctl.Command class definitely in the JAR. Don't know where to dig further.

Next code works fine - my relay turns ON and OFF perfectly.
`
            Pin dc = GPIOBoard.ofPin(72, PinDirection.OUTPUT);
            for (int i=0; i < 2; i++)
            {
                 dc.write(PinState.HIGH);
                Thread.sleep(1000);
                dc.write(PinState.LOW);
                Thread.sleep(1000);
            }
`
Thanks a lot in advance!

DigitalSmile, 2025-03-30:
> @DigitalSmile , BTW could you open a discussion section in your "gpio" project? I have a question about your library and I2C protocol, but don't know how to contact you.

You can simply create an issue in gpio repo, let's discuss it separately.

DigitalSmile, 2025-09-27:
We have delivered the first version of FFM provider and merged it into `develop` branch. Please, create new issues if any problems raising.

closed 2025-09-27 by DigitalSmile