Pi4J/pi4j #455: GpioDDigitalInput may never start monitoring line events

ylexus, 2025-03-27:
Sometimes, when my application starts, I notice that it never receives events from a GPIOD based digital input. After analysis of a heap dump when this happens and the source code of `GpioDDigitalInput` I came into conclusion that this is most likely caused by the loop in monitorLineEvents() never executing, and this is because there is a race between the thread calling `initialize()` (later called "thread A") executing the plain write to  `this.inputListener` and the executor thread executing `monitorLineEvents()` (later called "thread E") doing the plain read of `this.inputListener`. If thread E is fast enough, it may not yet see the value of `this.inputListener` set by thread A, for two reasons:
1. Thread E starts executing the submitted task (the loop) immediately
2. Plain/unsafe reads and writes are used for this field

(2) is actually less relevant, because even if they are converted to volatile access, there is no guarantee that Thread A write happens before Thread E read.

The `FutureTask` state is "2" (NORMAL) - see attached screenshot of a heap dump - which proves that there was no unhandled exception and the loop exited normally, and the only condition it could exit normally is that thread E has seen `this.inputListener` as `null`.

The design needs to be changed a little. First, I can see that access to `inputListener` is plain/unsafe from all methods, even the public ones, which runs the risk of visibility issues. `inputListener` is likely to needed to be made `volatile`. Second, using this field as loop exit condition in `monitorLineEvents` looks completely redundant, because this field is only written once in initialise() and is set to a non-null value there, so the loop can't really ever see a non-null value of this field UNLESS it's a race described above which can't be how it's intended to work.

[image]

ylexus, 2025-03-27:
I've created a demo-style PR on how I suggest this is fixed. I also noticed there was a thread.sleep loop slowing down the shutdown, I fixed that too. I can's see any existing tests for this code so I did not change any tests.

ylexus, 2025-05-19:
I wonder when this will be released? 2 months is way too long...

FDelporte, 2025-05-20:
I see this was merged just after we released 3.0.1.
At this moment, these are the changes for a new release:
https://github.com/Pi4J/pi4j/compare/3.0.1...develop
@taartspi @eitch are you OK with a release 3.0.2?

FDelporte, 2025-05-20:
Fixed in version 3.0.2.

mpilone, 2025-08-24:
I just opened [478](https://github.com/Pi4J/pi4j/issues/478) that was likely introduced by this fix, or is another variation of this bug. It seems like the root issue in both cases is that the monitor loop relies on the future returned from starting the loop. That's a race condition that needs to be avoided.