Pi4J/pi4j #472: Improvement: Sample for AnalogInput

TobiasReich, 2025-07-26:
I tried to read from Analog Pins but just can't get things done.
Sadly there seems to be no sample for that in the documentation and the sample project seems hopelessly outdated.

When I tinkered a minimal sample I got the error:  _Pi4J provider IO type [ANALOG_INPUT] could not be found._ so I tried adding a _AnalogInputProvider_ analog to the _add(GpioDDigitalInputProvider.newInstance())_ without any success.

Perhaps you could update the documentation / samples project in this regard. It would help a lot.

Thanks!

FDelporte, 2025-07-30:
Hi @TobiasReich, indeed no analog GPIOs on the Raspberry Pi!
Alternative approach with ADS1115: https://www.pi4j.com/examples/components/joystickanalog/
Or this one by @taartspi: https://github.com/Pi4J/pi4j-example-devices/tree/master/src/main/java/com/pi4j/devices/ads1256

taartspi, 2025-07-30:
https://github.com/Pi4J/pi4j-example-devices/tree/master/src/main/java/com/pi4j/devices/      I have an example MCP3008 ADC, I think it was cheap,  certainly a lot cheaper than the ads1256