Pi4J/pi4j #473: Pi5 - PWM not working

TobiasReich, 2025-07-28:
Using pi4j on my Pi5 I followed the sample from this site:
https://www.pi4j.com/blog/2024/20240423_pwm_rpi5/

However when running it I get the following error:

```
[main] ERROR com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm - /sys/class/pwm/pwmchip2/npwm
java.nio.file.NoSuchFileException: /sys/class/pwm/pwmchip2/npwm
        at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:106)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
        at java.base/sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:261)
        at java.base/java.nio.file.Files.newByteChannel(Files.java:379)
        at java.base/java.nio.file.Files.newByteChannel(Files.java:431)
        at java.base/java.nio.file.Files.readAllBytes(Files.java:3268)
        at java.base/java.nio.file.Files.readString(Files.java:3346)
        at java.base/java.nio.file.Files.readString(Files.java:3305)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.internal.LinuxPwm.getChannels(LinuxPwm.java:126)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.internal.LinuxPwm.channels(LinuxPwm.java:114)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.initialize(LinuxFsPwm.java:75)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.initialize(LinuxFsPwm.java:47)
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRuntimeRegistry.add(DefaultRuntimeRegistry.java:95)
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRegistry.add(DefaultRegistry.java:97)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl.create(LinuxFsPwmProviderImpl.java:99)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl.create(LinuxFsPwmProviderImpl.java:47)
        at com.pi4j@3.0.2/com.pi4j.context.Context.create(Context.java:355)
        at com.pi4j@3.0.2/com.pi4j.internal.IOCreator.create(IOCreator.java:58)
        at com.pi4j@3.0.2/com.pi4j.internal.IOCreator.create(IOCreator.java:116)
        at com.pi4j.example/com.pi4j.example.MinimalExample.initGPIO(MinimalExample.java:115)
        at com.pi4j.example/com.pi4j.example.MinimalExample.main(MinimalExample.java:50)
Error in initGPIO Failed to initialize IO PWM-2
java.lang.IllegalStateException: Failed to initialize IO PWM-2
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRuntimeRegistry.add(DefaultRuntimeRegistry.java:102)
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRegistry.add(DefaultRegistry.java:97)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl.create(LinuxFsPwmProviderImpl.java:99)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl.create(LinuxFsPwmProviderImpl.java:47)
        at com.pi4j@3.0.2/com.pi4j.context.Context.create(Context.java:355)
        at com.pi4j@3.0.2/com.pi4j.internal.IOCreator.create(IOCreator.java:58)
        at com.pi4j@3.0.2/com.pi4j.internal.IOCreator.create(IOCreator.java:116)
        at com.pi4j.example/com.pi4j.example.MinimalExample.initGPIO(MinimalExample.java:115)
        at com.pi4j.example/com.pi4j.example.MinimalExample.main(MinimalExample.java:50)
Caused by: com.pi4j.exception.InitializeException: Unable to get number of pins/channels supported by PWM chipset for PWM [2] @ </sys/class/pwm/pwmchip2>; /sys/class/pwm/pwmchip2/npwm
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.initialize(LinuxFsPwm.java:80)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.initialize(LinuxFsPwm.java:47)
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRuntimeRegistry.add(DefaultRuntimeRegistry.java:95)
        ... 8 more
```

I used the settings given in the sample project (i.e.  it is set to _dtoverlay=pwm_)
The code compiles without errors.
And the provider is there:

```
ugin.gpiod.provider.gpio.digital.GpioDDigitalInputProviderImpl} 
├─PWM: [1] <com.pi4j.io.pwm.PwmProvider> 
│ └─PROVIDER: "LinuxFS PWM Provider" {linuxfs-pwm} <com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl> {com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmP
```
The console output for the pwm classes shows chip0 and 1 though. However none is chip2 that failed in the Exception:

```
$ ls /sys/class/pwm/
pwmchip0  pwmchip1
```

I'm executing the program from Visual Studio Code without any run parameters. The default is mapped to 2 but even when I change this value in line 38 to 

`int address = 1;`

It still tries to map eventually on pwmchip2 as shown below: (so this doesn't seem the place to configure it propperly).

```
[main] ERROR com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm - /sys/class/pwm/pwmchip2/npwm
java.nio.file.NoSuchFileException: /sys/class/pwm/pwmchip2/npwm
        at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:106)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
        at java.base/sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:261)
        at java.base/java.nio.file.Files.newByteChannel(Files.java:379)
        at java.base/java.nio.file.Files.newByteChannel(Files.java:431)
        at java.base/java.nio.file.Files.readAllBytes(Files.java:3268)
        at java.base/java.nio.file.Files.readString(Files.java:3346)
        at java.base/java.nio.file.Files.readString(Files.java:3305)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.internal.LinuxPwm.getChannels(LinuxPwm.java:126)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.internal.LinuxPwm.channels(LinuxPwm.java:114)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.initialize(LinuxFsPwm.java:75)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwm.initialize(LinuxFsPwm.java:47)
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRuntimeRegistry.add(DefaultRuntimeRegistry.java:95)
        at com.pi4j@3.0.2/com.pi4j.registry.impl.DefaultRegistry.add(DefaultRegistry.java:97)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl.create(LinuxFsPwmProviderImpl.java:99)
        at com.pi4j.plugin.linuxfs@3.0.2/com.pi4j.plugin.linuxfs.provider.pwm.LinuxFsPwmProviderImpl.create(LinuxFsPwmProviderImpl.java:47)
        at com.pi4j@3.0.2/com.pi4j.context.Context.create(Context.java:355)
        at com.pi4j@3.0.2/com.pi4j.internal.IOCreator.create(IOCreator.java:58)
        at com.pi4j@3.0.2/com.pi4j.internal.IOCreator.create(IOCreator.java:116)
        at com.pi4j.example/com.pi4j.example.MinimalExample.initGPIO(MinimalExample.java:115)
        at com.pi4j.example/com.pi4j.example.MinimalExample.main(MinimalExample.java:50)
Error in initGPIO Failed to initialize IO PWM-1
java.lang.IllegalStateException: Failed to initialize IO PWM-1
```

So where is this pwmchip2 coming from and how can I change it to 0 or 1 I have. (And related to this: why do I have these channels with 
```

[all]
dtoverlay=pwm
```

written in the config?

FDelporte, 2025-07-30:
Indeed PWM can be a bit challenging...

Did you follow this?
https://www.pi4j.com/documentation/io-examples/pwm/#raspberry-pi-5

This is a working example I created myself:
https://github.com/Pi4J/pi4j-jbang/blob/main/Pi4JPWMLed.java

taartspi, 2025-07-30:
I was looking at this last night.  Yes pwmchip2 is not in the device tree no matter how u edit the config.sys. Was months back last time is used pwm on my pi5.   Reading the raspberry os support forum. Others see the same problem and the thought it us some update thar broke this.  Then they discuss what to add to fix the problem.  I did not find a clear answer and have some time to resume this afternoon

FDelporte, 2025-07-30:
Related info found by @DigitalSmile:

https://pi4j.slack.com/archives/C01J6P9TSH1/p1753881932816099
_I found the reason: I flashed Raspberry Pi OS again with the rpi-imager tool. This image uses kernel 6.6. Without any further updates and just activating the overlay, the pwmchip2 appears. After an update, it disappears. The issue seems to be related to the newest update to kernel 6.12
https://raspberrypi.stackexchange.com/questions/151149/pwmchip2-not-appearing-in-sys-class-pwm-on-raspberry-pi-5_

https://github.com/raspberrypi/linux/issues/6818 
the discussions in rpi kernel repo

taartspi, 2025-07-30:
@FDelporte and @DigitalSmile found additional posts to what I read last night and it confirms the with newer updates the device tree has chip 0 as the PWM.      
Using dtoverlay=pwm-2chn,pin=12,func=4,pin2=13,func2=4 and echo command I confirmed the gpios work,   
 I have not yet read if this is a permanent change although I would bet yes it is.    In the mean time I will cobble together what you can add into your program to proceed and then decide if modify the linuxfs PWM provider.

TobiasReich, 2025-07-30:
> Related info found by [@DigitalSmile](https://github.com/DigitalSmile):
> 
> https://pi4j.slack.com/archives/C01J6P9TSH1/p1753881932816099 _I found the reason: I flashed Raspberry Pi OS again with the rpi-imager tool. This image uses kernel 6.6. Without any further updates and just activating the overlay, the pwmchip2 appears. After an update, it disappears. The issue seems to be related to the newest update to kernel 6.12 https://raspberrypi.stackexchange.com/questions/151149/pwmchip2-not-appearing-in-sys-class-pwm-on-raspberry-pi-5_
> 
> [raspberrypi/linux#6818](https://github.com/raspberrypi/linux/issues/6818) the discussions in rpi kernel repo

Someone there asked "if you have 0 and 1 what do you need 2 for?". Well it leads me to the original thought. Is there a chance I can change this? At least changing the "channel" variable in the sample project didn't seem to be changing anything. At least for now I could live with using another channel as replacement if I would know how to do this.

taartspi, 2025-07-30:
I completed the linuxfs PWM      The channel usage is  : channel 0 would map to pwm0, not pwmchip0 .     
ls /sys/class/pwm/pwmchip0/
device  export  npwm  power  pwm0  subsystem  uevent  unexport

TobiasReich, 2025-07-31:
> I completed the linuxfs PWM The channel usage is : channel 0 would map to pwm0, not pwmchip0 . ls /sys/class/pwm/pwmchip0/ device export npwm power pwm0 subsystem uevent unexport

I'm not 100% sure how you did that but I understand you found a solution?
If that's the case please let me know: how could I solve this issue?

taartspi, 2025-07-31:
Thanks for the channel change. And channel/address mean the same thing in existing 
Pmw config This is an example of cleanup we can do now. Only channel should remain.    It will be this weekend before i can get back to this.

TobiasReich, 2025-08-04:
Wow, thanks for the quick response.
With this added to the java file I get another error:

```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.13.0:compile (default-compile) on project pi4j-example-minimal: Compilation failure
[ERROR] /home/pi/projects/java-projects/pi4j-colorLED-PWM/src/main/java/com/pi4j/example/MinimalExample.java:[7,40] package com.pi4j.plugin.linuxfs.provider.pwm is not visible
[ERROR]   (package com.pi4j.plugin.linuxfs.provider.pwm is declared in the unnamed module, but module com.pi4j.example does not read it)
[ERROR] ...
```

I know I'm a noob with Maven (using gradle mostly but the sample did everything in Maven so I gave it a try). Looks like I have to fix the module-info or so. Gonna let you know once I solved this issue.

taartspi, 2025-08-05:
I tried the 'patch' and my Pi5 pwm works with pwmchip0.   FYI. I used the simple     
``` 
dtoverlay=pwm

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on     
```   
Turned off audio to be certain it did not try to use GPIO18,   the code used channel/address  value 2     
I still need to properly fix the linuxfs-pwm provider so it selects the correct device path.  Not sure when a real fix will be available, if this is a problem for you please speak up