Pi4J/pi4j #488: Writing to SPI without device connected logs an error -- but does not throw

stefanhaustein, 2025-09-07:
I'd expect failed writes to throw if no device is connected at the requested bus/address, but I only get this message:

```
[main] ERROR com.pi4j.plugin.linuxfs.provider.spi.LinuxFsSpi - Could not write SPI message. ret -1, error: 90
```

IAmNickNack, 2025-09-14:
How does the Pi know that no device is connected?

IO is essentially fire and forget. Failure to send/receive would require some ACK/NACK to exist at the protocol level. SPI doesn't do this.

IAmNickNack, 2025-09-15:
I suppose `LinuxFsSpi` is simply echoing the behaviour of the native library in this case and we get the same behaviour as if we were writing C and calling `ioctl` directly. 

Adding an exception to the API in this case may make sense for the Pi4j abstraction, but it would be a breaking change to users that might be handling this based on the `-1` returned by `transfer`. (https://github.com/Pi4J/pi4j/blob/c41b3b37fe7d414797d4c71a2c553cd2165278c9/plugins/pi4j-plugin-linuxfs/src/main/java/com/pi4j/plugin/linuxfs/provider/spi/LinuxFsSpi.java#L281-L282)

There is a possible issue here in that `LinuxFsSpi` doesn't actually stick the the interface definition in `Spi.java`, which appears to indicate that a C-style error code will be returned and not the number of bytes transferred (I think those mentioned in the javadoc are actually Pigpio error codes too, so maybe not relevant to the pi 5):

https://github.com/Pi4J/pi4j/blob/c41b3b37fe7d414797d4c71a2c553cd2165278c9/pi4j-core/src/main/java/com/pi4j/io/spi/Spi.java#L125

I wouldn't expect a breaking change to be implemented as part of a minor revision, but this would really be up to the maintainers. 

In the meantime, if you'd prefer an exception, you could wrap the the Spi instance in your own implementation, delegate calls to `read`, `write` and `transfer` and throw an exception when `-1` is returned.

stefanhaustein, 2025-09-21:
Is this behaviour consistent with older pi4j SPI implementations? It seems certainly surprising to me and not consistent with how writing to streams in Javas normally behaves.

IAmNickNack, 2025-09-27:
I noticed that the `develop` branch appears now to be `4.0.0-snapshot`. This could be a good time to raise a PR with a breaking change which might resolve this issue

DigitalSmile, 2025-09-27:
I bet we need to add more validations before running any hardware calls.