Pi4J/pi4j #500: Pi4J.newAutoContext() no longer works in 4.0.0-SNAPSHOT build for gpiod

IAmNickNack, 2025-09-27:
On Pi5 I previous builds, `newAutoContext` worked. `3.1.0-SNAPSHOT` didn't have this problem.

`GpioDContext` appears not to mind `chipName` being null. `DefaultContextBuilder` now defaults `gpio.chip.name` to `unknown` and causes any chip not to be found.

https://github.com/Pi4J/pi4j/commit/55bead2b4f32e981d0aae51d1bdf72e8b671302e#diff-9d62545857e9677be136acfdc496571edced749bea4f39aeef2a7e6695723c5fR307

```
java.lang.IllegalStateException: Couldn't find gpiochip with name: unknown
	at com.pi4j.library.gpiod.internal.GpioDContext.initialize(GpioDContext.java:132) ~[pi4j-library-gpiod-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalInputProviderImpl.initialize(GpioDDigitalInputProviderImpl.java:48) ~[pi4j-plugin-gpiod-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalInputProviderImpl.initialize(GpioDDigitalInputProviderImpl.java:14) ~[pi4j-plugin-gpiod-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.provider.impl.DefaultRuntimeProviders.initializeProvider(DefaultRuntimeProviders.java:273) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.provider.impl.DefaultRuntimeProviders.add(DefaultRuntimeProviders.java:249) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.provider.impl.DefaultRuntimeProviders.add(DefaultRuntimeProviders.java:229) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.provider.impl.DefaultRuntimeProviders.initialize(DefaultRuntimeProviders.java:354) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.runtime.impl.DefaultRuntime.initialize(DefaultRuntime.java:321) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.context.impl.DefaultContext.<init>(DefaultContext.java:113) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.context.impl.DefaultContext.newInstance(DefaultContext.java:76) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.context.impl.DefaultContextBuilder.build(DefaultContextBuilder.java:319) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.context.impl.DefaultContextBuilder.build(DefaultContextBuilder.java:49) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
	at com.pi4j.Pi4J.newAutoContext(Pi4J.java:70) ~[pi4j-core-4.0.0-SNAPSHOT.jar:na]
```

IAmNickNack, 2025-09-27:
It looks like the a change was made to the common code to allow this not to fail in the FFM providers: 

```java
        var chipName = context.config().properties().get("gpio.chip.name");
        var digitalOutput = new DigitalOutputFFM(chipName, this, config);
```

https://github.com/Pi4J/pi4j/blob/develop/plugins/pi4j-plugin-ffm/src/main/java/com/pi4j/plugin/ffm/providers/gpio/DigitalOutputFFMProviderImpl.java#L23

eitch, 2025-09-27:
Do you need gpiod, or can you just remove the library and see if it works? FFM should become the default with 4.0

IAmNickNack, 2025-09-27:
I don't necessarily need gpiod, no. 

Cool that FFM will become the default. What's the plan for gpiod? Is it going to deprecated in favour of FFM or is it going to be removed entirely?

eitch, 2025-09-27:
I totally agree. It's not decided when we remove the old libraries,  but version 3.x is the stable one to use,  4.x is JDK25 with FFM and other features.

I would really like to remove all modules with native compiling to remove the use of make, gcc, docker etc. for building.  But as i said nothing is decided yet.

IAmNickNack, 2025-09-27:
Any plan to remove native code would remove LinuxFS too?

stefanhaustein, 2025-09-28:
Regardless of the deprecation / removal state of gpiod, can we please make sure that autocontext works? 

Lots of the examples have explicit linuxfs references, and I think this is how such stuff creeps in... :-/

stefanhaustein, 2025-09-28:
Could it be that https://github.com/Pi4J/pi4j/pull/501 is just hiding the problem now? 

I don't get an error message any longer, but either SPI or DigitalOutput seems to silently fail now with an autocontext on a raspery pi5 at head.

stefanhaustein, 2025-09-28:
I am using whatever autocontext is giving me -- which I expected to be something stable and reliable?

stefanhaustein, 2025-09-28:
Adding ffm to the POM makes it show up in the log, but then later it can't be found when trying to reference it explicitly:

```
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.context.impl.DefaultContext - Detected board model: Raspberry Pi 5 Model B
[main] INFO com.pi4j.context.impl.DefaultContext - Running on: Name: Linux, version: 6.12.47+rpt-rpi-2712, architecture: aarch64
[main] INFO com.pi4j.context.impl.DefaultContext - With Java version: Version: 25, runtime: 25+36-LTS, vendor: Azul Systems, Inc., vendor version: Zulu25.28+85-CA
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Initializing Pi4J context/runtime...
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider DIGITAL_OUTPUT RaspberryPi Digital Output (GPIO) Provider (raspberrypi-digital-output) with priority 0 with provider GpioD Digital Output (GPIO) Provider (gpiod-digital-output) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider DIGITAL_INPUT RaspberryPi Digital Input (GPIO) Provider (raspberrypi-digital-input) with priority 0 with provider GpioD Digital Input (GPIO) Provider (gpiod-digital-input) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_OUTPUT LinuxFS Digital Output (GPIO) Provider (linuxfs-digital-output) with priority 100 as lower priority than GpioD Digital Output (GPIO) Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider PWM RaspberryPi PWM Provider (raspberrypi-pwm) with priority 0 with provider LinuxFS PWM Provider (linuxfs-pwm) with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider SPI RaspberryPi SPI Provider (raspberrypi-spi) with priority 0 with provider LinuxFS SPI Provider (linuxfs-spi) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_INPUT LinuxFS Digital Input (GPIO) Provider (linuxfs-digital-input) with priority 100 as lower priority than GpioD Digital Input (GPIO) Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider I2C RaspberryPi I2C Provider (raspberrypi-i2c) with priority 0 with provider LinuxFS I2C Provider (linuxfs-i2c) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider SPI FFM API Provider SPI (ffm-spi) with priority 1 as lower priority than LinuxFS SPI Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider PWM FFM API Provider PWM (ffm-pwm) with priority 1 as lower priority than LinuxFS PWM Provider which has priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider SERIAL RaspberryPi Serial Provider (raspberrypi-serial) with priority 0 with provider FFM API Provider Serial (ffm-serial) with higher priority 1
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_OUTPUT FFM API Provider Digital Output (ffm-digital-output) with priority 1 as lower priority than GpioD Digital Output (GPIO) Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider I2C FFM API Provider I2C (ffm-i2c) with priority 1 as lower priority than LinuxFS I2C Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_INPUT FFM API Provider Digital Input (ffm-digital-input) with priority 1 as lower priority than GpioD Digital Input (GPIO) Provider which has priority 150
[main] INFO com.pi4j.library.gpiod.internal.GpioDContext - Using chip gpiochip0 pinctrl-rp1
[main] INFO com.pi4j.library.gpiod.internal.GpioDContext - GpioD context already initialized with chip: gpiochip0
[main] INFO com.pi4j.platform.impl.DefaultRuntimePlatforms - adding platform to managed platform map [id=raspberrypi; name=RaspberryPi Platform; priority=5; class=com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform]
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Pi4J context/runtime successfully initialized.
com.pi4j.provider.exception.ProviderNotFoundException: Pi4J provider [ffm-digital-output] could not be found.  Please include this 'provider' JAR in the classpath.
```

stefanhaustein, 2025-09-30:
@IAmNickNack did your patch fully resolve the problem for you? Are you using gpio in and out in your setup?

mores, 2025-09-30:
Just my 2 cents. gpio in is working for me:

`[INFO] Running com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - setUp
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected OS: Name: Linux, version: 6.12.34+rpt-rpi-2712, architecture: aarch64
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected Java: Version: 25, runtime: 25+36-LTS, vendor: Azul Systems, Inc., vendor version: Zulu25.28+85-CA
[main] INFO com.pi4j.boardinfo.util.BoardInfoHelper - Detected board type MODEL_5_B by code: d04170
WARNING: A restricted method in java.lang.System has been called
WARNING: java.lang.System::load has been called by com.pi4j.library.gpiod.util.NativeLibraryLoader in an unnamed module (file:/home/smore/.m2/repository/com/pi4j/pi4j-library-gpiod/4.0.0-SNAPSHOT/pi4j-library-gpiod-4.0.0-SNAPSHOT.jar)
WARNING: Use --enable-native-access=ALL-UNNAMED to avoid a warning for callers in this module
WARNING: Restricted methods will be blocked in a future release unless native access is enabled
[main] INFO com.pi4j.context.impl.DefaultContext - Detected board model: Raspberry Pi 5 Model B
[main] INFO com.pi4j.context.impl.DefaultContext - Running on: Name: Linux, version: 6.12.34+rpt-rpi-2712, architecture: aarch64
[main] INFO com.pi4j.context.impl.DefaultContext - With Java version: Version: 25, runtime: 25+36-LTS, vendor: Azul Systems, Inc., vendor version: Zulu25.28+85-CA
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Initializing Pi4J context/runtime...
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider DIGITAL_INPUT RaspberryPi Digital Input (GPIO) Provider (raspberrypi-digital-input) with priority 0 with provider GpioD Digital Input (GPIO) Provider (gpiod-digital-input) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider DIGITAL_OUTPUT RaspberryPi Digital Output (GPIO) Provider (raspberrypi-digital-output) with priority 0 with provider GpioD Digital Output (GPIO) Provider (gpiod-digital-output) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider SPI RaspberryPi SPI Provider (raspberrypi-spi) with priority 0 with provider LinuxFS SPI Provider (linuxfs-spi) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_OUTPUT LinuxFS Digital Output (GPIO) Provider (linuxfs-digital-output) with priority 100 as lower priority than GpioD Digital Output (GPIO) Provider which has priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider PWM RaspberryPi PWM Provider (raspberrypi-pwm) with priority 0 with provider LinuxFS PWM Provider (linuxfs-pwm) with higher priority 100
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Replacing provider I2C RaspberryPi I2C Provider (raspberrypi-i2c) with priority 0 with provider LinuxFS I2C Provider (linuxfs-i2c) with higher priority 150
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Ignoring provider DIGITAL_INPUT LinuxFS Digital Input (GPIO) Provider (linuxfs-digital-input) with priority 100 as lower priority than GpioD Digital Input (GPIO) Provider which has priority 150
[main] INFO com.pi4j.library.gpiod.internal.GpioDContext - Using chip gpiochip0 pinctrl-rp1
[main] INFO com.pi4j.library.gpiod.internal.GpioDContext - GpioD context already initialized with chip: gpiochip0
[main] INFO com.pi4j.platform.impl.DefaultRuntimePlatforms - adding platform to managed platform map [id=raspberrypi; name=RaspberryPi Platform; priority=5; class=com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform]
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Pi4J context/runtime successfully initialized.
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - testOne
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - DI1: com.pi4j.io.gpio.digital.impl.DefaultDigitalInputConfig@6bd61f98
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880Driver - seesaw chipId: -1
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - Set Red
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - Pos: 0
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - Set White
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - Pos: 0
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - Set Blue
[main] INFO com.pi4j.drivers.input.rotary.adafruit5880.Adafruit5880DriverTest - Pos: 0
Position1: 1
Position2: 1
Position1: 2
Position2: 2
`

stefanhaustein, 2025-09-30:
This is the exception I am seeing for gpiod in v4:

```
Exception in thread "Timer-0" com.pi4j.io.exception.IOException: Failed to set state for output DOUT-25 to LOW
	at com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalOutput.state(GpioDDigitalOutput.java:97)
	at com.pi4j.io.gpio.digital.DigitalOutputBase.off(DigitalOutputBase.java:276)
	at com.pi4j.io.gpio.digital.DigitalOutputBase.off(DigitalOutputBase.java:44)
	at com.pi4j.drivers.display.graphics.st7789.St7789Driver.command(St7789Driver.java:110)
	at com.pi4j.drivers.display.graphics.st7789.St7789Driver.command(St7789Driver.java:116)
	at com.pi4j.drivers.display.graphics.st7789.St7789Driver.setPixels(St7789Driver.java:159)
	at com.pi4j.drivers.display.graphics.GraphicsDisplay$DriverEntry.transferBuffer(GraphicsDisplay.java:330)
	at com.pi4j.drivers.display.graphics.GraphicsDisplay$DriverEntry.transferBuffer(GraphicsDisplay.java:277)
	at com.pi4j.drivers.display.graphics.GraphicsDisplay.transferBuffer(GraphicsDisplay.java:252)
	at com.pi4j.drivers.display.graphics.GraphicsDisplay.flush(GraphicsDisplay.java:113)
	at com.pi4j.drivers.display.graphics.GraphicsDisplay$1.run(GraphicsDisplay.java:236)
	at java.base/java.util.TimerThread.mainLoop(Timer.java:572)
	at java.base/java.util.TimerThread.run(Timer.java:522)
Caused by: com.pi4j.library.gpiod.internal.GpioDException: c_gpiod_line_set_value failed: -1 (Operation not permitted)
	at com.pi4j.library.gpiod.internal.GpioD.lineSetValue(GpioD.java:256)
	at com.pi4j.library.gpiod.internal.GpioLine.setValue(GpioLine.java:118)
	at com.pi4j.plugin.gpiod.provider.gpio.digital.GpioDDigitalOutput.state(GpioDDigitalOutput.java:95)
	... 12 more
```

DigitalSmile, 2025-10-01:
Closing as resolved.

eitch, 2025-10-01:
I see you solved the issue. So yes i managed to test and see that all the providers with exception of FFM for I2C worked for my tests. I think we need to be careful about context etc. Perhaps multi-threading and opening/closing of I/Os during application life time can cause problems. I'm sure this hasn't been tested too carefully yet.