Pi4J/pi4j #508: FFM PermissionHelper doesn't work on PI5 Bookworm

IAmNickNack, 2025-09-27:
File permission checks fail on my device. 

PermissionHelper assumes the `gpio` group on Debian which fails on what is (to my knowledge) the default setup on Pi5

```
➜  ~ uname -a
Linux rpi-5 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
➜  ~ ls -l /dev/gpiochip0
crw-rw---- 1 root dialout 254, 0 Sep 27 13:17 /dev/gpiochip0
➜  ~ ls -l /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Sep 27 13:17 /dev/i2c-1
➜  ~ ls -l /dev/spidev0.0
crw-rw---- 1 root spi 153, 0 Sep 27 13:17 /dev/spidev0.0
```

Additionally, the helper is applied to i2c and SPI, but doesn't check the device type and assumes these should also be in the `gpio` group.

https://github.com/Pi4J/pi4j/blob/develop/plugins/pi4j-plugin-ffm/src/main/java/com/pi4j/plugin/ffm/common/PermissionHelper.java#L103-L112

FFM appears to work once the check is removed. 

Should the check be fixed or removed?

IAmNickNack, 2025-09-27:
Would a simpler read-write check work?

stefanhaustein, 2025-10-01:
Doe I understand this correctly? 

- On one hand, the code is only needed to educate the user -- but not essential for operation. 
- On the other hand, it requires some platform / os dependend calls that are tricky to get right everywhere.

If that's the case, could we put it into a try/catch block -- at least as a workaround until there is a proper fix?

stefanhaustein, 2025-10-01:
I think a workaround would simplify debugging https://github.com/Pi4J/pi4j/issues/514

With different parts of v4 in flux / under development IMHO, we should really have a focus on unblocking developers

eitch, 2025-10-02:
@stefanhaustein i agree with you, but you can just add a skip in the relevant method, build it and you're good to go =). That is what i did when i ran into the issue yesterday.

stefanhaustein, 2025-10-02:
Sorry by no means I meant to ask for rushing for a proper fix. 

My problem here is that there is an obvious workaround (try/catch with printing a stack trace) that would 

- enable you to take all the time needed to address this properly without rush
- unblock others immediately without a hack

People hacking around checks is dangerous because they might not remove these hacks and then code won't get the early testing it deserves.

DigitalSmile, 2025-10-04:
> People hacking around checks is dangerous because they might not remove these hacks and then code won't get the early testing it deserves.

But the opposite, commit workarounds to main/develop branch is also not a good way to go. It is not a release, it is a snapshot for testing. Bugs happens. We are here to fix them, no rush.

stefanhaustein, 2025-10-04:
> But the opposite, commit workarounds to main/develop branch is also not a good way to go. 

I work at a relatively large software company with a monorepro and the only reason why you would get agreement there for this statement is because the "good way" would be rolling back the offending change.

Commonly accepted priority of handling breaking changes there is:

1. Roll back the offending change to stop blocking others and get back to a well-known working state. Re-submit a fixed version later.
2. Fix-forward or work around immediately. Frowned upon because it might lead to even more rollbacks
3. Leave the code in a broken state until a proper fix can be found. Just added for reference, I have never seen this happening.

I guess one reason why I am getting a bit irritated here is that this gets so much engrained in development culture that it feels absolutely absurd to treat breakages any other way.

Of course we are not at a large company here and "we are all working on this in our private/free time". 

That said, it's not just your private/free time. It's also the private/free time of others who depend on core for their own development. 

I have made other arguments for a workaround above that I think address your stated concerns better than leaving the current state as-is and I won't repeat them here.

Stefan

DigitalSmile, 2025-10-04:
> I work at a relatively large software company with a monorepro

Let us not fall into the argument "which company I am working is bigger and has monorepo approach to development". This is childish.

> That said, it's not just your private/free time. It's also the private/free time of others who depend on core for their own development.

Again, I will point out - this is not release, just a snapshot and testing phase is in process. If you see any issues with current snapshot you are always free to downgrade to previous stable version to continue your development.

> I have made other arguments for a workaround above that I think address your stated concerns better than leaving the current state as-is and I won't repeat them here.

I agree, let's stop this argument, there is no point of wasting internet bytes on this conversation.

closed 2025-10-06 by eitch