Pi4J/pi4j #509: FFM PWM provider configuration is incompatible with the LinuxFS version

IAmNickNack, 2025-09-28:
The same configuration cannot be used between FS and FFM PWM providers.

As a user I would expect to be able to use the same configuration, regardless of the underlying provider. 

With the FS provider I can use the following configuration. Admittedly, this relies on the FS plugin defaulting the chip to `0`):

```kotlin
    // LinuxFS
    val pwm = pi4j.create(
        Pwm.newConfigBuilder(pi4j)
            .id("refresh")
            .name("Refresh PWM")
            .address(2)                   // PWM channel
            .pwmType(PwmType.HARDWARE)
            .frequency(220)
            .dutyCycle(50)
            .shutdown(0)
            .build()
    )
```

The FFM provider changes the definition of `address` to now refer to the chip and uses `busNumber` instead to refer to the channel. 

```kotlin
    // FFM
    val pwm = pi4j.create(
        Pwm.newConfigBuilder(pi4j)
            .id("refresh")
            .name("Refresh PWM")
            .address(0)                   // PWM chip
            .pwmType(PwmType.HARDWARE)
            .frequency(220)
            .dutyCycle(50)
            .shutdown(0)
            .busNumber(2)                 // PWM channel
            .build()
    )
```

This seems contradictory to the plugin architecture. From the user's point of view these are now different device types requiring bespoke configurations.

---

The FFM version does make sense as it makes the configuration explicit by removing a dependency on properties if a different chip is required. 

A fix might not revert the FFM version to be compatible with FS. Maybe the FS version needs a breaking change in V4 to make it compatible with FFM and maintain the usability of the plugin architecture.

IAmNickNack, 2025-09-28:
I see. So another consequence of the addressing issue here? => https://github.com/Pi4J/pi4j/issues/481

FDelporte, 2025-09-29:
This indeed calls for a review of address and/or busnumber in the different types of protocols. At this moment, it's not clear to me what the best solution could be, but being on V4.0.0-SNAPSHOT does allow us to introduce some breaking changes... 
What are some suggestions to fix this?

IAmNickNack, 2025-09-29:
My 2 cents:

The issue arrises because the same terms are used to mean different things depending on the provider, but the unique constraint doesn't/can't take this into account. It's implemented at a level in the object graph where any required metadata isn't available.

This "metadata" only becomes available at the provider level. Each individual provider knows what makes devices of its type unique. **Maybe this should be a provider responsibility (?)**. 

Moving the responsibility of the provider might be a little more complicated to implement across the digital-in and digital-out providers as they would need to share some state. (This might raise the question of whether two providers are actually needed for GPIO 🧌 )

What is/was the aim of enforcing uniqueness on `address`? If it's to avoid assigning the same pin in multiple places, I don't think this is really possible/reliable between providers using `address` because a GPIO pin can't be so easily compared to a PWM or SPI device, etc.

* is a unique constraint required/helpful?
* is a constraint more relevant/achievable/reliable at a provider level?
* is the `address` terminology used in the API appropriate to all devices?

---
Refer back to [this issue](https://github.com/Pi4J/pi4j/issues/481#issuecomment-3243174503)

FDelporte, 2025-09-29:
Indeed the goal was to prevent the same pin to be used incorrectly for different uses, but @eitch already improved this by checking on both address and IOType here:
https://github.com/Pi4J/pi4j/commit/bad88b6da550cd478d21d63da6d2542fc4f4d883

IAmNickNack, 2025-09-29:
I've not tested it, but I think @taartspi is suggesting that it could be an issue if you want multiple PWM channels on the same chip. 

If this is an issue, I think it illustrates how the uniqueness is might be provider specific.

DigitalSmile, 2025-09-30:
Can we make a stop here to design better API for the user?
> GPIO : address = pin number
> SPI and I2C: address = address within the SPI/I2C space
> PWM = chip + address
> Serial: address = string of the /dev/...

That's where we get the trouble - confusing API.
So what should be done:
- make specific attributes for all available interfaces (e.g. for GPIO it should be explicitly `pin` instead of `address`). This should be done in core and we need to adjust all providers. It could be a breaking change, but we can support two types of config at the same time, just make legacy one `@Deprecated`
- make sure you can create different channels/pins/etc on one bus, where applicable
- make a slight better optional API without using auto-context, like in example here https://github.com/Pi4J/pi4j/tree/develop/plugins/pi4j-plugin-ffm/src/main/java/com/pi4j/plugin/ffm/api Usually you definitely know what board you are using and hardware pins are not changed in a while. That means we can create a bunch of helpers with explicit methods calls, like `spi1(...)` or `pwm0(Channel.ONE)` so that end user can easily use them without knowing all underlying stuff with providers/configs/plugins/ids/etc
- make decision with auto-context and default provider to be used

Calling @FDelporte @taartspi @eitch to the thread

eitch, 2025-09-30:
I like your idea @DigitalSmile I'm thinking that we really should accept that with 4.x we're doing breaking changes. We did this also with 2.x but now that so much is going on, and we are forcing JDK25 and other things we should really talk about breaking changes and what this allows us to do. We will be supporting 3.x for a while and not deprecate it, so 4.x can be open to this. What does everyone think?

4.x is for FFM and JDK25 and thus could allow us to do this.
3.x is stable on all current platforms and thus people don't need to upgrade yet.